FreeCAD/FreeCAD #10364: [Problem] Creation of thumbnail file in exported .3mf file

andesfreedesign, 2023-08-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Devuan GNU/Linux 4 (chimaera) (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Spanish/Argentina (es_AR)
Installed mods: 
  * 3D_Printing_Tools
  * Assembly4 0.50.2
  * freecad.gears 1.0.0
  * CurvedShapes 1.0.4
  * MeshRemodel 1.8919.0
  * A2plus 0.4.60n
  * lattice2 1.0.0
  * Curves 0.6.13
  * Assembly3 0.12.0
  * fasteners 0.4.62
  * ExplodedAssembly
  * sheetmetal 0.2.63
  * DynamicData 2.46.0
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Problem description

I found a problem in the current version 0.21 regarding the creation of the thumbnail.png file in the Metadata folder of the exported .3mf files in FreeCAD.
Some time ago I developed a Thumbnailer for 3mf files and when I uploaded it to GitHub it worked...

https://github.com/andesfreedesign/Thumbnailer-3mf

Now, reviewing an exported .3mf file, I found that the thumbnail.png file was created empty...
On another computer I have FreeCAD_weekly-builds-33345-2023-06-25-conda-Linux-x86_64-py310.AppImage, I tried to export to 3mf and the thumbnailer works perfectly. Therefore, some change in the code, after that date (2023-06-25), must be causing the problem.

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2023-08-30:
This could also mean that off-screen rendering doesn't work on this computer. Does it work for you to create an image from an opened 3d view? You will find the command in the Tools menu.
Or if you activate the option to embed a thumbnail into FCStd projects does this work, too?

andesfreedesign, 2023-09-05:
> This could also mean that off-screen rendering doesn't work on this computer. Does it work for you to create an image from an opened 3d view? You will find the command in the Tools menu. Or if you activate the option to embed a thumbnail into FCStd projects does this work, too?

Sorry for the delay in answering...
Yes, both renderings work
I share a small video to show how it works

[image]

I estimate that when modifications were made due to the export problems to 3mf and its opening in a slicer, something changed in the code, now thumbnails are not generated.

Can it be in this file?

src/Mod/Mesh/Gui/ThumbnailExtension.cpp

wwmayer, 2023-09-05:
> src/Mod/Mesh/Gui/ThumbnailExtension.cpp

Yes, this is where the thumbnail is created.

wwmayer, 2023-09-06:
When looking at the code then there is one special thing: the background is transparent
With the "Save image" function you have this option, too. Does it work to create an image if you set the background to transparent? 
Hint 1: You have to choose an image format that supports transparencies like PNG.
Hint 2: Clicking on the button Extended shows you the further options

luzpaz, 2023-09-24:
> When looking at the code then there is one special thing: the background is transparent
With the "Save image" function you have this option, too. Does it work to create an image if you set the background to transparent?
Hint 1: You have to choose an image format that supports transparencies like PNG.
Hint 2: Clicking on the button Extended shows you the further options

@andesfreedesign can you confirm ?

luzpaz, 2023-10-13:
@andesfreedesign ping

luzpaz, 2023-11-26:
@andesfreedesign please respond

luzpaz, 2023-12-16:
@andesfreedesign

closed 2025-02-08 by github-actions[bot]