FreeCAD/FreeCAD #10369: bug

a937983423, 2023-08-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: EndeavourOS (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33985 (Git)
Build type: None
Branch: master
Hash: 602841c5f0f2c9ce332f9d7c8df7153ac73454ba
Python 3.11.3, Qt 5.15.10, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: Chinese/China (zh_CN)
Installed mods: 
  * kicadStepUpMod 10.19.4
  * InventorLoader 1.4.0
  * CurvedShapes 1.0.4
  * EM
  * AirPlaneDesign 0.4.0
  * Behave-Dark-Colors 0.1.1
  * DesignSPHysics 0.6.3(24-09-2022)
  * Dracula 0.0.5
  * Design456 0.0.1
  * 3DfindIT 1.2.0
  * Estimate 0.1.2
  * DynamicData 2.46.0
  * dodo 1.0.0
  * A2plus 0.4.60n
  * Cfd
  * 3D_Printing_Tools
  * CubeMenu
  * ExplodedAssembly
  * Beltrami 1.0.8-alpha
  * Assembly4 0.50.2
  * BCFPlugin 1.0.0
  * Alternate_OpenSCAD 1.0.0
  * fasteners 0.4.60
  * CfdOF 1.24.4
  * Curves 0.6.13
  * ArchTextures
  * sheetmetal 0.2.63
  * AnimationFreeCAD 1.0.0
  * freecad.gears 1.0.0
  * FEM_FrontISTR 0.1.0
  * Fcmcua 1.0.0
  * Defeaturing 1.2.0
[/code]
```

### Subproject(s) affected?

Addon Manager

### Problem description

15:23:05  QOpenGLFunctions created with non-current context
15:23:11  flag
15:23:11  flag2
15:23:12  environment var 'WM_PROJECT_DIR' is not defined
,
              fallback to default /opt/openfoam5
15:23:12  environment var 'WM_PROJECT_VERSION' is not defined, fallback to hard-coded (5, 0)
15:23:12  PyFoam installatoin path: /usr/lib/python3.11/site-packages/PyFoam/__init__.py
15:23:18  (qt.qpa.wayland) Wayland does not support QWindow::requestActivate()

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-08-26:
@a937983423 can you removed the *cfd* workbench from the addon manager or manually by deleting it in your local .FreeCAD/ config directory and see if you can reproduce the issue?

@oliveroxtoby pardon, have you seen this issues with the cfdof wb before ?

luzpaz, 2023-08-26:
Note: this maybe related to #10048

oliveroxtoby, 2023-08-28:
> ~@a937983423 can you removed the _cfd_ workbench from the addon manager or manually by deleting it in your local .FreeCAD/ config directory and see if you can reproduce the issue?~ Nvm try this from your shell:
> 
> ```shell
>  XDG_SESSION_TYPE=x11 path/to/freecad
> ```
> 
> @oliveroxtoby pardon, have you seen this issues with the cfdof wb before ?

No, I haven't seen this. As you noted, the output shown seems to be coming predominantly from CFD rather than CfdOF though.

adrianinsaval, 2023-09-21:
`QOpenGLFunctions created with non-current context`
and
`(qt.qpa.wayland) Wayland does not support QWindow::requestActivate()`

are unrelated to cfdOF, they are present even with a clean config, but only happen when running on a wayland session.

@wwmayer are these errors that we should be filtering out or are they relevant? The one about requestActivate happens many times during usage of the app.

I also get this error when looking at the notifications things:
`(qt.qpa.wayland) setGrabPopup called with a parent, QtWaylandClient::QWaylandXdgSurface(0x5593945d8e40) which does not match the current topmost grabbing popup, QtWaylandClient::QWaylandXdgSurface(0x55939512f580) According to the xdg-shell protocol, this is not allowed. The wayland QPA plugin is currently handling it by setting the parent to the topmost grabbing popup. Note, however, that this may cause positioning errors and popups closing unxpectedly because xdg-shell mandate that child popups close before parents`

wwmayer, 2023-09-21:
> are these errors that we should be filtering out or are they relevant?

I have no idea if they indicate a deeper problem or not. At least they can be suppressed by adding
`stream << "qt.qpa.wayland.warning=false\n";`
to setCategoryFilterRules in Application.cpp

Andrei-Pozolotin, 2024-03-13:
note: 
* as of `0.22.0dev.36405` - `FreeCAD` on `Wayland` is unusable 
* `ArchLinux/KDE` has now switched to `Wayland` by default, making this issue unavoidable
* workaround with launcher: `XDG_SESSION_TYPE=x11 /usr/bin/FreeCAD` sort of, barely, works on `QT5`

```
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36405 (Git)
Build type: None
Branch: main
Hash: 0d228bcec48ccd4c0112beb48695b6c53e58f9a6
Python 3.11.8, Qt 5.15.12, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
```

closed 2025-02-08 by github-actions[bot]

nmschulte, 2025-04-15:
Has any effort been put toward resolving this?  Perhaps it's something Qt6 helps improve?

I've been instructing Qt5 to use the `xcb` platform plugin, as it used to fail with the `wayland` variants (`wayland-egl`/`libqwayland-egl.so, `wayland`/`libqwayland-generic.so`); I forget the message now it's been so long.

```
FreeCAD: ./src/glue/gl.cpp:2249: const cc_glglue* cc_glglue_instance(int): Assertion `current_ctx && "Must have a current GL context when instantiating cc_glglue!! (Note: if you are using an old Mesa GL version, set the environment variable COIN_GL_NO_CURRENT_CONTEXT_CHECK to get around what may be a Mesa bug.)"' failed.
Aborted
```

However, now, I find that FreeCAD seems to work seemingly the same if I use `QT_QPA_PLATFORM=wayland` vs `xcb` in addition to `COIN_GL_NO_CURRENT_CONTEXT_CHECK=1`.

```
QOpenGLFunctions created with non-current context
```
```
Coin warning in glxglue_isdirect(): Couldn't get current GLX context.
```
```
This plugin supports grabbing the mouse only for popup windows
```