FreeCAD/FreeCAD #10371: [Problem] Cyclic reference abusively reported for Bodies in a Part

FlachyJoe, 2023-08-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33981 (Git)
Build type: Unknown
Branch: master
Hash: 5a153e50ffb6046b6cc4dba498344032f6edb952
Python 3.11.4, Qt 5.15.10, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: French/France (fr_FR)
```

### Subproject(s) affected?

PartDesign

### Problem description

### Description
In some conditions a cyclic reference is reported when trying to SubShapeBind a shape of a Body in another one and they are children of a same Part.

### Steps to reproduce
1. Open a new document
2. Create a Std_Part
3. Create a PartDesign_Body into
4. Create a Box in this body
5. Create a second PartDesign_Body in the Std_Part
6. Try to create a SubShapeBinder of a face of the box
7. **Get the message "Cyclic reference to Unnamed#Part"**

### Workarounds
1. Create Std_Part
2. **Create the 2 bodies**
3. Activate Body
4. Create a Box
5. Activate Body001
6. Create a SubShapeBinder of a face of the Box
7. It works !

Or

1. Open a new document
2. Create a Std_Part
3. Create a PartDesign_Body into
4. Create a Box in this body
5. **Create a SubShapeBinder of a face of the box in this first Body**
6. Create a second PartDesign_Body in the Std_Part
7. Create a SubShapeBinder of a face of the box
8. It works !

### Anything else?

https://forum.freecad.org/viewtopic.php?t=80624

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kisolre, 2023-11-19:
Further discussion at [https://forum.freecad.org/viewtopic.php?t=82817](url)

Andrei-Pozolotin, 2023-12-15:
@realthunder : 
 
the issue is still present as of:

```
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35391 (Git)
Build type: None
Branch: makepkg
Hash: ed10fe216d332debfd317d4639e284c50f5dab47
Python 3.11.6, Qt 5.15.11, Coin 4.0.1, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)

```

PR #11750 (2024-01-08): Fix object path in active object when creating new body in App::Part
Fixes #10371

The circular reference is caused by incomplete object path in active object (`Gui.activeView().getActiveObject('pdbody', True)`) when creating new body in an active App::Part.
commit 50fc8e86 (2024-01-08): PD: fix creating new body in active part

Add to active App::Part before setting active body in order to obtain
correct object path when setting active body

Fixes #10371
closed 2024-01-08 by sliptonic