FreeCAD/FreeCAD #10372: [Problem] FreeCAD does not have a privacy policy

chennes, 2023-08-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
0.21.0
```

### Subproject(s) affected?

None

### Problem description

An elementary school district in the United States reached out to the FreeCAD Forums administration email address yesterday to inquire about a privacy policy for the software itself (not for the forums, blog, etc.). It is common for school districts to require a written policy so that they can ensure software is protecting student privacy. We should add one, keeping in mind that we do have an embedded Chromium-based web browser, and we allow you to embed your name in your FCStd files.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2023-08-26:
We could have a privacy policy that says just that... 

1) the software does not collect any private data nor does it send anything over the internet, 
2) some modules (web, start, addonmgr...) do allow internet access, but those can be disabled (this should be a link to where we describe how)
3) the files you save with FreeCAD might contain private data such as the name you configured in the preferences, or file paths if you use features that rely on external files

Additional question, how can an administrator lock internet access of FreeCAD and not permit other users to enable it

grubertechnic, 2023-08-26:
> 2\. some modules (web, start, addonmgr...) do allow internet access, but those can be disabled (this should be a link to where we describe how)

As FOSS, this should be disable as default and opt in, not opt out, even when it is simple browser.

chennes, 2023-08-26:
> As FOSS, this should be disable as default and opt in, not opt out, even when it is simple browser.

It is -- if you do a clean install of FreeCAD you are asked whether you want to enable internet access. We do need a way for an administrator to prevent this opting-in, however.

grubertechnic, 2023-08-26:
> It is -- if you do a clean install of FreeCAD you are asked whether you want to enable internet access.

When was this implemented? Around 2 weeks ago i updated 2 machines from Debian 11 to 12 and make everything new, from new git repo pull to new fresh compile, even i delete all configs (but only on one machine). But i never get asked anything. Also sometimes for testing i clean the configs, then i also never get asked.

chennes, 2023-08-26:
Did you try to get Start to access the internet? I think that's when you get asked.

grubertechnic, 2023-08-26:
Arg, i switch off web and start in cmake .... forget about it, my mistake.

FOSCODER, 2023-08-28:
Schools won't really care about BIM library tool and maybe addon manager. but web and start are a problem.

Zolko-123, 2023-08-29:
> An elementary school district in the United States reached out ...

this is very anecdotal, we shouldn't do anything about such single user comments. If the same question comes up frequently **THEN** we should look into it, but not for every random request. If a school want's to limit Internet access for its pupils then it has to do so with its own network administrator. But having an open network in the school and asking FreeCAD to close that network should be a no-go : this is not our business.

yorikvanhavre, 2023-08-29:
Beyond this particular case I think we need to provide a way to make FreeCAD not requiring ANY connection to the internet. There are many reasons and use cases for that. It's not a huge change, it would not hinder existing users, but it should be possible to have a FreeCAD install that is guaranteed to not access the internet

chennes, 2024-06-05:
Really, it can be closed because we did add a privacy policy. Thanks!

closed 2024-06-05 by chennes