FreeCAD/FreeCAD #10374: [Problem] Auto-Ranging in PART isn't helping

fester225, 2023-08-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: macOS 13.5.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Problem description

When using the PART workbench, and moving a group across any axis, in Imperial (sorry), after the group, or object, has been moved more than 12 inches, the system changes to incrementing a foot at a time. After three feet, it increments a yard at a time.

This might be helpful for other users, but not for me. 

Would it be possible to include an option to turn off this auto-ranging feature? 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-08-26:
In the Part WB there is no move function. How do you move it, be explicit! Now it is pure guesswork.

grubertechnic, 2023-08-26:
Besides that, what unit system you using? On imperial decimal in its not jumping.

fester225, 2023-08-26:
I'm using imperial decimal. 
I forgot to include the problem occurs in PART/PLACEMENT.

Sorry.

grubertechnic, 2023-08-26:
There is no Part WB Placement, there is a general Placement, but anyway, no issue with imperial decimal, see

[2023-08-26 18-48.webm](https://github.com/FreeCAD/FreeCAD/assets/109620914/ea05270f-33c8-41c1-9f16-5b80ac939d9f)

```
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33990 (Git)
Build type: Release
Branch: master
Hash: d623e37e5ef269848c979915ed480ea9bbf7ce83
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
```

fester225, 2023-08-26:
Thank you. I'll have to find a way to use screen shot to take movies to document the problem.

fester225, 2023-08-30:
https://github.com/FreeCAD/FreeCAD/assets/5368100/adae1b52-d02d-4f1c-9be4-eebfc04cb486

grubertechnic, 2023-08-31:
Still works for me as expected. Try the weekly 0.22 build, could be, that the behavior was added past 0.21 release, because also the `"` will be transformed for me after scrolling the value automatic to `in`.

Roy-043, 2023-08-31:
Forum topic:
https://github.com/FreeCAD/FreeCAD/issues/10374

luzpaz, 2023-11-26:
@JohnOCFII can you reproduce on macOS?

@fester225 can you try the latest development version https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds and report back ?

maxwxyz, 2024-06-13:
I cannot reproduce and will close this issue as there is no feedback.

closed 2024-06-13 by maxwxyz