FreeCAD/FreeCAD #10375: [Problem]

fester225, 2023-08-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: macOS 13.5.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Problem description

PART Workbench: When moving a group, the system displays the new position of the group. However, the components of the group are shown to be wherever they were originally created.

Would it be possible to change the way MOVE works so all the components inside the group are re-positioned at the same time? 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-08-26:
> Would it be possible to change the way MOVE works so all the components inside the group are re-positioned at the same time?

Use for that the appropriate tool, Part Container. Group is only for logical grouping, not for positioning.

@luzpaz : please consider it for closing.

FEA-eng, 2023-08-26:
Can you add some meaningful title to the [Problem] label (e.g. "[Problem] Part group components aren't repositioned when move is used on the group") ?

closed 2023-08-27 by luzpaz

fester225, 2023-08-30:
Where is PART CONTAINER? 
It is not in the PART workbench.

grubertechnic, 2023-08-31:
https://wiki.freecad.org/Std_Part

Please read the manual or ask question in the forum, this is a bug tracker, no helpdesk.