FreeCAD/FreeCAD #10381: [Problem] StartPage has a number of visual styling issues

qewer33, 2023-08-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
OS: Fedora Linux 38 (KDE Plasma) (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Glass
  * lattice2 1.0.0
  * kicadStepUpMod 10.19.4
  * Manipulator 1.5.0
  * Assembly4.backup1691951820.225657 0.50.2
  * Assembly4
  * Behave-Dark-Colors 0.1.1
  * Curves 0.6.13
  * DynamicData 2.46.0
  * fasteners 0.4.59
  * Freecad-Built-in-themes-beta 1.2.2
  * freecad.gears.backup1692223246.5590403 1.0.0
  * freecad.gears
  * ExtremeProLight 1.6.0
  * A2plus 0.4.60n
  * ExplodedAssembly
  * FreeCAD_DarkPreferencePAck 0.0.1
```

### Subproject(s) affected?

Other (specify in description)

### Problem description

The StartPage has a number of issues on it's CSS styling and visuals. Take a look at the list below:

- The drop shadow of the tabs is inconsistent with other shadows and is hardcoded
[image]

- Some elements are centered while some elements are left-aligned, visually inconsistent
[image]

- The width of the page seems to be set with a percentage value so it's outer padding/margin changes when the size of the window or the central area changes
[image]

- Panels on the Activity tab don't fill full width
[image]

- The main panel shouldn't change size, it should fill full width and height and it's content should scroll (instead of the whole page)
[image]

- The scrollbar should be styled based on preference values (QtWebEngine uses Chromium afaik and chromium scrollbar is styleable)

### Anything else?

I'd like to open a PR for the StartPage that addresses these issues and modernizes it's look without going too far off from the current design and layout. Below is a mockup:

[image]

This proposed PR *will not remove any of the current StartPage features*. It will only change the visual styling of the StartPage. I would be happy to hear suggestions (or further styling issues to fix) before I start working on this.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

obelisk79, 2023-08-26:
I like all of the proposed adjustments. Of note, the tab glow effect has driven me nuts for a while, this would be a refreshing change.

Syres916, 2023-08-28:
@PLChris requested that the icons be adjustable in size on top of the original described changes.

qewer33, 2023-08-28:
@PLChris screenshot of the changes in the above PR

[image]

[image]

closed 2023-08-28 by qewer33