FreeCAD/FreeCAD #10393: [Problem] Draft "Array" doesn't work after i change it's name

Mackowaty007, 2023-08-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
[/code]
```

### Subproject(s) affected?

Draft

### Problem description

After creating a body i went into the "Draft" workspace and made a 2D array of that body. Everything works fine, unless i change the name of that "Array" into anything else. Then whatever value i change it doesn't apply
[image]
(in the photo above i set the NumberZ to 4 but it doesn't apply because i changed the name to "testtest")

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-08-28:
The model needs to be recomputed.
Please read:
https://wiki.freecad.org/Std_Refresh
https://wiki.freecad.org/Tree_view#Overlay_icons

Roy-043, 2023-08-28:
Closing this as completed.

closed 2023-08-28 by Roy-043