FreeCAD/FreeCAD #10396: [Path] estlcam post processor broken

mlampert, 2023-08-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Path

### Problem description

Opening the preferences for the Path -> Job Preferences -> Post Processor and moving the mouse over the estlcam entry in the "Post Processor" list produces the following error:
```
Traceback (most recent call last):                                                                 
  File "/home/markus/projects/FreeCAD/mlampert/build/Mod/Path/Path/Main/Gui/PreferencesJob.py", line 346, in setProcessorListTooltip                                                                  
    self.setPostProcessorTooltip(self.form.postProcessorList, item.text(), "")                     
  File "/home/markus/projects/FreeCAD/mlampert/build/Mod/Path/Path/Main/Gui/PreferencesJob.py", line 339, in setPostProcessorTooltip                                                                  
    processor = self.getPostProcessor(name)                                                        
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                        
  File "/home/markus/projects/FreeCAD/mlampert/build/Mod/Path/Path/Main/Gui/PreferencesJob.py", line 333, in getPostProcessor                                                                         
    processor = PostProcessor.load(name)                                                           
                ^^^^^^^^^^^^^^^^^^^^^^^^                                                           
  File "/home/markus/projects/FreeCAD/mlampert/build/Mod/Path/Path/Post/Processor.py", line 47, in load                                                                                               
    exec("import %s as current_post" % postname, namespace)                                        
  File "<string>", line 1, in <module>                                                             
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import                                                                                          
    return original_import(name, *args, **kwargs)                                                  
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                  
  File "/home/markus/projects/FreeCAD/mlampert/build/Mod/Path/Path/Post/scripts/estlcam_post.py", line 30, in <module>                                                                                
    import PathScripts.PostUtils as PostUtils                                                      
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import                                                                                          
    return original_import(name, *args, **kwargs)                                                  
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                  
ModuleNotFoundError: No module named 'PathScripts.PostUtils' 
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-01:
CC @haraldhartmann

haraldhartmann, 2023-09-01:
> CC @haraldhartmann

I will fix it next week after my vacation.

luzpaz, 2023-09-23:
@haraldhartmann ping

PR #10789 (2023-09-25): Path: refactor Utils imports in estlcam_post.py
fix #10396
commit 3bb891ea (2023-09-25): fix #10396 - refactor Utils imports
closed 2023-09-25 by sliptonic