FreeCAD/FreeCAD #10397: [Problem] Issue with Transformation tool on a Body

aperezhra, 2023-08-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: macOS 13.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * dodo 1.0.0
  * Curves 0.6.13
[/code]
```

### Subproject(s) affected?

None

### Problem description

While attemting to move (transform) a Body to a new position using the Transformation tool from the right-click menu on the Tree view or using the Edit->Transform tool the object moves on the scree, but the Body returns back to it original display position despite the new Placemente coordinates:

Steps to recreate:
**1. Body#1 (for reference)**
2. Create a circle sketch
4. Pad the sketch, 5mm
**5. Body#2 (to Transform)**
6. Create a link from sketch from Body#1
7. Create a sub-object shape binder from the linked Sketch
8. Pad the binder sketch, 10mm
9. Transform the Body (either through Right-click on the tree view or from the Edit>Transform option)
11. Move the Object to a new position on the X or Y axis
12. Click Ok
**13. The Body#2 returns to its original display position, even though the Body's Placement Position has been changed.**

[image]

### Anything else?

This is the same issue reported by #8886 for version 0.20 but closed without a solution.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-08-28:
What you are seeing in due to (intended) behavior of the Binder. It is controlled by the Relative property of the Binder. You need to set it to false.

Roy-043, 2023-08-30:
Closing this.