FreeCAD/FreeCAD #10404: Update on vers 0.21

sebaras, 2023-08-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
By updating from Drawings (TechDraw), from FC 0.20.2 to FC 21, positions are lost.
Also lost are the dimmension of lines from Baloons with Square boxes.

Eveb not Cleasr ist how the Complex Section works. please add on Youtube a film, how it works.

Thanks
```

### Subproject(s) affected?

Techdraw

### Problem description

By updating from Drawings (TechDraw), from FC 0.20.2 to FC 21, positions are lost.
Also lost are the dimmension of lines from Baloons with Square boxes.

Eveb not Cleasr ist how the Complex Section works. please add on Youtube a film, how it works.

Thanks 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-08-28:
This issue lacks just about everything.
Clear title?
Version info?
Example file?
Step-by-step instructions?

You need to improve it.

Roy-043, 2023-08-30:
This issue still lacks too much to be usable. Therefore closing this.