FreeCAD/FreeCAD #10406: [Problem] TechDraw: No Zoom possible

tud69, 2023-08-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 7 build 7601
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Assembly4 0.12.5
[/code]
```

### Subproject(s) affected?

Techdraw

### Problem description

Zooming does not work in TechDraw workbench. The menu items are greyed out, the keyboard shortcuts CTRL+/- or SHIFT/CTRL+ don't work either. There is literally no way to zoom on a laptop. The only solutions is to buy / borrow a mouse with wheel - making it impossible to work on the train or airplane...

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-08-28:
What navigation model are you using?  I do all my FreeCAD work on a laptop using Touchpad navigation.  No mouse required.

I don't use the zoom from the menu or the keyboard shortcuts, so I've never noticed that they don't work.   Thanks for pointing this out.

Roy-043, 2023-08-28:
Zooming with keyboard shortcuts does work (even though they are greyed out in the menu). But the TechDraw page does have to have the keyboard focus. Activating a page's tab or clicking on the page in the main view area should accomplish that.

These all work:
Ctrl++ (numpad)
Ctr+- (numpad)
Ctr+- (standard keyboard)
Ctrl+Shift++ (standard keyboard)

tud69, 2023-08-28:
Well, it does not! I am going through all this pain, signing up for a GitHub account, specifically just to report this bug.

Of course the window and tab have the keyboard focus. Plus, is there a any good reason why those menu items should be greyed out otherwise?

@WandererFan: You are right. Changing navigation to TouchPad, the PageUp/PageDown buttons do work for zooming. Thanks!
@Roy-043 : This wouldn't be possible if the window wouldn't have to focus. 

So this is still a bug.

[image]

Roy-043, 2023-08-29:
With CAD navigation, and after making sure the page in the main view area has the keyboard focus, the 4 shortcuts I have mentioned work. They also work for the 3D view.

With TouchPad navigation the 4 mentioned shortcuts only work for the 3D view. As you have already found, PageUp/PageDown must then be used for a TechDraw page. PageUp/PageDown then also works for the 3D view, but the zoom behavior is different from the Ctrl shortcuts (different zoom factor and zoom center).

So there is some room for improvement here:
* Clarify that the menu items only apply to the 3D view or make them work in TechDraw as well.
* Unifying PageUp/PageDown and Ctrl zoom behavior in the 3D view.
* Find out why Ctrl zoom is disabled in TD with TouchPad navigation. I do not see a good reason ATM.

WandererFan, 2023-08-30:
Unless you have a numeric keypad, CTRL++ is really CTRL+Shift++.   CTRL+Shift++ seems to work for all the modes I've tried so far, except Touchpad.

tud69, 2023-08-31:
Again, I need to repeat myself:

Please remember we are in TechDraw workbench!

CTRL - works
CTRL + does not

CTRL/SHIFT/- does nothing
CTRL/SHIFT/+ does nothing

This is true for navigation modes CAD, TouchPad, Blender and probably the rest as well (am I supposed to test all of them?). True in the environment as stated on top.

Look, I am getting tired of this now... if you guys don't believe me and I have to prove each one of you that I am not stupid.... I leave this up to you now. I have better things to do with my limited time.

Thanks for the otherwise awesome FreeCAD !

grubertechnic, 2023-08-31:
Works here too flawless.

```
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34091 (Git)
Build type: Release
Branch: master
Hash: a2735416e1189566329b327d99f115ced7071b86
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
```

But as @Roy-043 mentioned, it need the focus that it works. As i can see in the screenshot, you are in the PartDesign WB and want to zoom in TechDraw. It does not mean, that this does not work, but at least it need a focus on the screen/sheet, else some key combination can have different meanings, when have to focus on different things.

> Clearly a bug

I see no bug here, but also again as @Roy-043 mentioned, room for improvements. It is in general a good thing, as in the forum before adding things in the tracker.

> I have better things to do with my limited time.

WE ALL have only very limited time for working on it.

Roy-043, 2023-08-31:
@tud69's different experience could perhaps be related to the keyboard layout. Maybe the left and right CTRL key behave differently? It could also be that the CTRL++ shortcut is caught by the OS.

tud69, 2023-08-31:
OK, last comment here.
@grubertechnic First screenshot was taken in the wrong workbench (my bad), 2nd screenshot was correct in the TechDraw WB. Doesn't change the fact that it behaves the way as stated in my previous post where zooming in doesn't work.

"But as @Roy-043 mentioned, it need the focus that it works. As i can see in the screenshot, you are in the PartDesign WB and want to zoom in TechDraw. It does not mean, that this does not work, but at least it need a focus on the screen/sheet, else some key combination can have different meanings, when have to focus on different things."

Blah blub... sorry. How do you explain that CTRL- works then, if the sheet wasn't focused? Did you read all my previous posts CAREFULLY? Tired of this, seriously. READ CAREFULLY FOLKS !

@Roy-043 I have German keyboard layout on the laptop, no num pad present.

Good bye everyone! Keep up the good work!

WandererFan, 2023-08-31:
I believe you.  If I change my keyboard layout to German (using the first entry of many German layouts?), neither ctl+ or ctl- work in the modes I tried.  

Both work in any mode other than Touchpad with US or Canadian Multilingual layouts.

I'll have to investigate what key the system thinks I am pressing with the German layout.

grubertechnic, 2023-09-01:
I have a german keyboard layout, it works, just said, but i am on a different OS.

WandererFan, 2023-09-04:
@tud69, is this the keyboard you use? 
[image]

If so, do you use the keys outlined in red or blue as +/-?  The blue keys are +/- in the QWERTY KB, but are not reported as +/- if I set my kb layout to German.  The red keys are +/- in the QWERTZ kb and are reported as +/-.

The red keys do zoom in/out with my layout set to German.

luzpaz, 2023-10-09:
What's left to do here ?

WandererFan, 2023-10-09:
Close it unless somebody with a German KB reports it again.   AFAICT +/- zooming generally works.

closed 2023-10-09 by WandererFan