FreeCAD/FreeCAD #10408: [Problem] Keyword arguments are not available in python commands

AjinkyaDahale, 2023-08-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
N/A
```

### Subproject(s) affected?

Core

### Problem description

When running python commands that use C++ in the background, for example in `DocumentObject.addProperty`, some arguments are optional, but since they are defined by keywords, if the `n`th argument needs to be provided, all the previous `n-1` arguments also need to be provided, regardless of whether or not they are needed (or even optional).

### Anything else?

TODO: Create a list of commands where this applies.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-08-28:
Shouldn't:
'since they are defined by keywords'
Be:
'since they are NOT defined by keywords'
?

AjinkyaDahale, 2023-08-28:
@Roy-043 thanks. Fixed.