FreeCAD/FreeCAD #10411: Cannot create an arc with a large radius

mattkamsler, 2023-08-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

When I do the following:

1) Start a new FreeCad project
2) Change to the sketcher workbench
3) Create a new sketch on the XY-Plane
4) Create a new arc (doesn't matter which option I select center and end points or end points and rim points)
5) Add a radius constraint on the arc of 901 m.
6) Use fit all to zoom out
7) Attempt to move the arc.

What I see in the console is:
12:04:13  Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Not able to move point with the id and type: (0, 0)
12:04:13  App.getDocument('Unnamed').getObject('Sketch').movePoint(0,0,App.Vector(31228.166016,327418.750000,0),0)
12:04:13  Drag curve: Not able to move point with the id and type: (0, 0)

I am no longer able to fully constrain this arc in any way. For example if I attempt to fully constrain this arc in any way by adding distance, point constraints, and reference lines with angle constraints, I get the same error as above. Issues seem to start somewhere around 900 m.

### Anything else?

The issue was brought up as part of this discussion: https://forum.freecad.org/viewtopic.php?p=702591 in the FreeCAD forum. I narrowed my initial description down to the above scenario to narrow in on and simplify the problem.

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-09-07:
I know this doesn't fix the issue but are you aware of https://github.com/HakanSeven12/Trails the author no longer resides on the forum but your field and his workbench do seem to have similarities. I'm guessing by the screenshots that the transport roads etc are scaled down in the workbench to stop issues like you have reported.

mattkamsler, 2024-08-26:
Still a bug today in version .21.2. I can't constrain an arc with a large radius in sketcher.

FEA-eng, 2025-06-08:
@maxwxyz This is still relevant. How about reopening the issue?

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.42095 (Git)
Build date: 2025/06/06 16:36:48
Build type: Release
Branch: main
Hash: 9c017b0dbf538c2570df7c3316355cdcdc3163f5
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

FEA-eng, 2026-01-28:
I still get this sometimes when dragging the endpoint:

```
09:47:46  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Failed to move geometries.
09:47:46  Unnamed#Sketch: App.getDocument('Unnamed').getObject('Sketch').moveGeometries([(0, 1)], App.Vector(-484931, 747930, 0))
09:47:46  ViewProviderSketch: Drag: Failed to move geometries.
```

or even this error right after applying this large radius constraint:

```
09:49:20  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Invalid constraint index: 0
09:49:20  Unnamed#Sketch: App.getDocument('Unnamed').getObject('Sketch').setDatum(0,App.Units.Quantity('901000.000000 mm'))
```

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260127 (Git shallow)
Build date: 2026/01/27 15:33:30
Build type: Release
Branch: (HEAD detached at 7bcaa76)
Hash: 7bcaa766de101b10c03b4a6f4fd09292068d5ef6
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```