FreeCAD/FreeCAD #10415: [Problem] Build failing on macOS with undeclared identifier

sfmissionmark, 2023-08-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
(anaconda3)markbutler@Marks-MacBook-Pro FreeCAD % git checkout 
Your branch is up to date with 'origin/master'.
```

### Subproject(s) affected?

None

### Problem description

/FreeCAD/FreeCAD/src/App/Application.cpp:2492:61: error: use of undeclared identifier 'FCVersionSuffix'
        App::Application::Config()["BuildVersionSuffix" ] = FCVersionSuffix;
                                                            ^
1 error generated.
ninja: build stopped: subcommand failed.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-08-28:
You need to recreate the Version.h file, either by doing a complete rebuild, or just deleting that single file from your build folder and letting the make system recreate it.

closed 2023-08-31 by adrianinsaval

luzpaz, 2023-09-01:
Do you know where the upstream ninja ticket is perhaps ?

adrianinsaval, 2023-09-01:
No but this specific file not getting updated was discussed before: https://github.com/FreeCAD/FreeCAD/discussions/9606