FreeCAD/FreeCAD #10422: [Problem] Changing anti-alias: no restart message?

benj5378, 2023-08-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33983 (Git) Snap 752
Build type: Release
Branch: master
Hash: 010dca83030357a1b9467e7c5843d51b103ecb70
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
```

### Subproject(s) affected?

Core

### Problem description

Okay, so I started my snap install of FreeCAD dev build. I changed the anti-alias... but nothing changed? Huh. I remembered the anti-aliasing being better than this... is it broken?

Well, I went to a (possibly older) dev build and tried the same. Now, when I changed the anti-alias, nothing happened either, but I got a message box saying, the anti-alias changes wouldn't go into effect until a restart of the program.

A restart showed to give the expected anti-alias on both my local build and the snap install of FreeCAD dev build. So why doesn't the newer version/snap version give the restart-needed message upon changing anti-alias preference, when it's needed?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-08-30:
FYI: instead of restarting you can also open a new 3D view to see the effects.

closed 2025-02-08 by github-actions[bot]

benj5378, 2025-02-08:
reopen

benj5378, 2025-04-05:
I still think this is an issue, could we please reopen @maxwxyz

luzpaz, 2025-11-25:
So the question: should we have a persistent warning for this each time with a checkbox option to ignore in the future ?

OR

Add a note in the preference explaining `*requires restart`