FreeCAD/FreeCAD #10423: [Problem]

ejvaldes, 2023-08-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * A2plus 0.4.60n
  * Assembly3 0.12.0
  * Assembly4 0.50.2
  * CurvedShapes 1.0.4
  * Curves 0.6.13
  * dxf-library
  * ExplodedAssembly
  * InventorLoader 1.4.0
  * sheetmetal 0.2.63
  * Ship 2022.4.11
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

When dimensioning, some dimensions hold other does not. Similarly while using dimensioning with spreadsheet data.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-08-29:
Besides there is no title, this issue is that vague formulated, that it is in fact useless. Pure guess, not knowing the CAD concept of constraints, see manual.

https://wiki.freecad.org/Sketcher_Workbench
https://wiki.freecad.org/Constraint

Please someone consider to close it, if there is no more information added.

Roy-043, 2023-08-30:
Closing this as the issue indeed lacks too much to be usable.

closed 2023-08-30 by Roy-043