FreeCAD/FreeCAD #10424: [Problem] Sketcher segfault on latest master

qewer33, 2023-08-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
OS: Fedora Linux 38 (KDE Plasma) (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34071 (Git)
Build type: Release
Branch: startpage-housekeeping
Hash: 932fd4638ba80671115fb2aa871ab3a273b6f673
Python 3.11.4, Qt 5.15.10, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Glass
  * lattice2 1.0.0
  * kicadStepUpMod 10.19.4
  * Manipulator 1.5.0
  * Assembly4.backup1691951820.225657 0.50.2
  * Assembly4
  * Behave-Dark-Colors 0.1.1
  * Curves 0.6.13
  * DynamicData 2.46.0
  * fasteners 0.4.59
  * Freecad-Built-in-themes-beta 1.2.2
  * freecad.gears.backup1692223246.5590403 1.0.0
  * freecad.gears
  * ExtremeProLight 1.6.0
  * A2plus 0.4.60n
  * ExplodedAssembly
  * FreeCAD_DarkPreferencePAck 0.0.1
```

### Subproject(s) affected?

Sketcher

### Problem description

1. Open the example file provided below in a latest master build (unzip first):
[PartDesignExample.zip](https://github.com/FreeCAD/FreeCAD/files/12465220/PartDesignExample.zip)

2. Try to open Sketch001 or Sketch003
3. FreeCAD crashes with a SEGFAULT

### Anything else?

Terminal output on crash:

<details>
<summary>Terminal Output</summary>
<pre>
FreeCAD 0.22.0, Libs: 0.22.0devR34071 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Failed to read /home/qewer/.local/share/FreeCAD/Mod/ExtremeProLight/package.xmlThe `.FCStd#` backup format is deprecated as of v0.21 and may be removed in future versions.
To update, check the 'Preferences->General->Document->Use date and FCBak extension' option.
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x3db70) [0x7f6775a5fb70]
[... 48 lines cut ...]
#49  /lib64/libQt5Widgets.so.5(+0x20d1a9) [0x7f677700d1a9]
#50  /lib64/libQt5Widgets.so.5(+0x21072f) [0x7f677701072f]
#51  0x7f6776faeb75 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib64/libQt5Widgets.so.5+0x85
#52  0x7f677b419b21 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/qewer/Desktop/Git/freecad-build/lib/libFreeCADGui.so+0xff
#53  0x7f67762b4188 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib64/libQt5Core.so.5+0x128
#54  0x7f677676cbf8 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /lib64/libQt5Gui.so.5+0x828
#55  0x7f677674acfc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib64/libQt5Gui.so.5+0xac
#56  /lib64/libQt5XcbQpa.so.5(+0x714d6) [0x7f676460f4d6]
#57  /lib64/libglib-2.0.so.0(g_main_context_dispatch+0x14c) [0x7f677331348c]
#58  /lib64/libglib-2.0.so.0(+0xba648) [0x7f6773371648]
#59  /lib64/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7f6773310b13]
#60  0x7f6776306729 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib64/libQt5Core.so.5+0x69
#61  0x7f67762b2b4b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib64/libQt5Core.so.5+0x13b
#62  0x7f67762baddb in QCoreApplication::exec() from /lib64/libQt5Core.so.5+0x9b
#63  0x7f677b2b4d9f in Gui::Application::runApplication() from /home/qewer/Desktop/Git/freecad-build/lib/libFreeCADGui.so+0x2c75
#64  ./FreeCAD() [0x42652d]
#65  /lib64/libc.so.6(+0x27b4a) [0x7f6775a49b4a]
#66  /lib64/libc.so.6(__libc_start_main+0x8b) [0x7f6775a49c0b]
#67  ./FreeCAD() [0x425725]
</pre>
</details>

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-08-29:
Confirmed. Newer files have no issue. Tried some older files, they crash also. Would be interesting, on which version of file it starts to crash.

chennes, 2023-08-29:
Here's the relevant part of the backtrace:
```
Sketcher::GeometryFacade::getGeometry<…>() const GeometryFacade.h:242
Sketcher::GeoListModel::getGeometryFromGeoId(const std::vector<…> &, int) GeoList.cpp:107
Sketcher::GeoListModel::getGeometryFromGeoId(int) const GeoList.cpp:134
SketcherGui::EditModeConstraintCoinManager::processConstraints(const Sketcher::GeoListModel<…> &) EditModeConstraintCoinManager.cpp:829
SketcherGui::EditModeCoinManager::processGeometryConstraintsInformationOverlay(const Sketcher::GeoListModel<…> &, bool) EditModeCoinManager.cpp:741
SketcherGui::ViewProviderSketch::draw(bool, bool) ViewProviderSketch.cpp:2990
SketcherGui::ViewProviderSketch::updateData(const App::Property *) ViewProviderSketch.cpp:3064
Gui::ViewProvider::update(const App::Property *) ViewProvider.cpp:308
Gui::ViewProviderDocumentObject::update(const App::Property *) ViewProviderDocumentObject.cpp:383
Gui::Document::slotChangedObject(const App::DocumentObject &, const App::Property &) Document.cpp:795
```

chennes, 2023-08-29:
Running a recompute object before loading the sketch seems to prevent the issue.

closed 2023-08-29 by chennes