FreeCAD/FreeCAD #10435: Draft: Draft_Offset: avoid OCC offset problems or handle them more gracefully

Roy-043, 2023-08-30:
Draft_Offset has an option to create OCC style offsets. But this option only works if the selected object is planar and has at least 2 edges. The code currenty does not verify this nor does it catch the OCC error.

Note that a single arc, although it does define a plane, also results in an OCC error.

Forum topic:
https://forum.freecad.org/viewtopic.php?t=80775

Roy-043, 2023-08-30:
The problem with the single arc is caused by Draft code.  In the case of circles and arcs the `dvec` argument of `offsets.offsetWire` receives a float instead of a vector. The function does not handle that. The `offsets.offsetWire` function is too complex IMO.

semhustej, 2024-10-14:
I encountered another small problem:

When trying to create Draft_Offset of **ellipse**, the report view informs user that `DraftGeomUtils: Couldn't project point` multiple times.
[image]

The proposed solution is to inform the user:
`Offset of this object type is not supported.`

------------------------
When trying to create Draft_Offset of **point**, there is no information in the report view.

The proposed solution is to inform the user:
`Offset of this object type is not supported.`

Roy-043, 2025-09-13:
It is still open AFAICT.