FreeCAD/FreeCAD #10436: [Problem] Sketcher tools need a taskbox

PaddleStroke, 2023-08-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
0.22 dev
```

### Subproject(s) affected?

Sketcher

### Problem description

The taskview is supposed to be a view showing settings of commands.
Yet in sketcher the taskview is used as a 'Widget area' and the sketcher tools don't have a TaskBox.

This is very limiting as the tool cannot have settings. Because of that, many tools are crippled and/or have inconsistent UIs using qmessage boxes. Some tools require multiple tools because of that : 
- copy/clone/move/rectangular-array.
- Fillet / Fillet preserving corner.
Everytime you want to add an option to a sketcher tool, as you can't add a setting to the toolbox, you have to either duplicate the tool or other bad solutions.

So besides being inconsistent with the rest of Freecad, it is crippling existing tool and further development.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sasobadovinac, 2023-08-30:
Well, maybe I don't fully understand your idea, but generally the taskbox is used for features to set the settings of the feature and this must be set when the feature is created because the feature itself is parametric.

sasobadovinac, 2023-08-30:
Ok, yes I think I see now the idea for it with arrays and offset...

PaddleStroke, 2023-08-30:
Here's for example the offset tool taskbox to better illustrate : 
[image]

sasobadovinac, 2023-08-30:
Yes, I was just reviewing this and was thinking to ask you why this solution from you would not be ok... So basically this is just a taks/issue for it or is it about something else?

sasobadovinac, 2023-08-30:
By the way, about the sketcher offset, I remember there were some discussions about the need to automatically create constraints for it... In an older version of Catia that I was working with some time ago, if I remember correctly, it was working like that that if the user selected a single element (for example one of the edges in the above rectangle) it would create all the needed constraints and construction geometry for it, but if teh users selected several connected elements (like your examples above) it would create them without any constraints (just like in the example above), not sure if in newer versions and in other cad's this is still working the same or different...

sasobadovinac, 2023-08-30:
> The solution from me (the screenshot above) is ok. I open this issue here to follow the contributing policy for the work on my 'tool-setting' branch on which we worked last year with Abdullah before the contribution process was a thing.

Yes this is ok, I guess the comparison to other taskboxes from other WBs, made me think it was about something else...

maxwxyz, 2023-10-28:
To understand it: for e.g. the offset tool I am able to set the value in the task box. Is the value displayed like a dimension in the sketch or can be edited later on or do I need to delete the geometry and create a new offset when I need to change the value?

Is it possible to offset referenced geometry from external edges in the body? (my most wanted use case)

PaddleStroke, 2023-10-28:
The created offset is not parametric. It is just sketcher geometries. The tool has an option to create a 'offset constraint' which is a collection of construction geometries and constraints that make the geometries act as if it was a parametric offset.
External geometries are currently not working, but we can make it work.

maxwxyz, 2023-10-28:
ok, so the offset constraint is what I am looking for, thanks 👍🏻 
In case of reference geometry I would prefer https://github.com/FreeCAD/FreeCAD/issues/10043 as this should then work automatically for offset?

PaddleStroke, 2023-10-28:
https://github.com/FreeCAD/FreeCAD/issues/10043 is a valid feature request. But offset should also accept external geos imo.

maxwxyz, 2023-10-28:
> #10043 is a valid feature request. But offset should also accept external geos imo.

does offset currently accept construction geometry? Than this could be a workaround if #10043 is implemented you could switch it to construction geo.

PaddleStroke, 2023-10-28:
Yes construction geos are working. https://github.com/FreeCAD/FreeCAD/issues/10043 would be a workaround yes. But I think we should have offset that works for external geos without workaround.

PR #11174 (2023-11-03): [Sketcher] Tool Parameter widget + Offset tool
Fixes #10436
Fixes #7221
Fixes #5839

This PR adds parameters to sketcher tools. Some settings on the 3d view directly. Some settings in a taskbox widget on the taskview.

This PR include : 
- Tool Parameter framework.
- Point tool
- Line tool
- Rectangle tool
- Offset tool
- Polygon tool
closed 2023-11-03 by abdullahtahiriyo