FreeCAD/FreeCAD #10437: [Problem] Navigation cube covers the top part of the legend in FEM

FEA-eng, 2023-08-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33992 (Git)
Build type: Release
Branch: master
Hash: 90bafb3a069468455d8c65d030ef8750b510c39b
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
[/code]
```

### Subproject(s) affected?

FEM

### Problem description

For a few releases, the navigation cube in the default location has been covering the top part of the legend in FEM results. It needs to be manually moved each time. The legend (or its labels) should be moved more to the left.

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-08-31:
I've made a small step in that if the NavigationCube is already activated then the ColorBar moves inbound: https://github.com/FreeCAD/FreeCAD/compare/master...Syres916:FreeCAD:Dev_Move_ColorBar_According_To_NavCube
But, if the user activates the NavCube after the ColorBar is created then at present it doesn't automatically adjust inbound which should be possible. It'll probably need triggering from the Preference Page.

With regard to the text color and size in your other Problem 10439, that is proving a proper stumbling block, really confusing code for my brain cells, there's plenty of color line codes which I've tried changing manually to prompt finding the correct one but to no avail as yet.

luzpaz, 2023-08-31:
Didn't donovaly work on this at some point ?

Syres916, 2023-08-31:
Not aware of anybody working on it in the past. Another approach I might look at tomorrow is to slightly shrink the colorbar in it's original location.

FEA-eng, 2023-08-31:
@Syres916 Thank you for taking a look at this issue. Many users complain about it. It's a quite annoying issue. The FEM workbench would look much more professional with a good-looking, well-visible legend.

@luzpaz I've only seen a forum discussion and PRs related to an issue that with some window sizes/resolutions, the labels of the legend were partially covered by the side bar of the FreeCAD's window.

Syres916, 2023-09-01:
I'm getting to hit another brick wall so I'm going to take a break for a day or so. Here's the current screenshot dealing with both issues, FC opened up as a brand new user with no Preferences set other than Touchpad nav mode:

[image]

I tried black as default but to be honest found it harder to read than white at the same size and font. I will look at how much work it'll be in to make it Stylesheet aware but the priority is the preferences setting which is proving difficult in the existing task panel, though it shouldn't be but I seem to find all these 'special' setups.

FEA-eng, 2023-09-02:
It already looks so much better. White is indeed the best as the default color and there just needs to be an option to change it if the background is also white, for instance (common when taking screenshots for reports).

Maybe legend settings could go to Preferences --> General ? But the most intuitive place is definitely that context menu of the legend, if there's enough space in it.

Syres916, 2023-09-02:
After a lot of gnashing of teeth I ended up putting both the label text color and text size in `Preferences --> Display --> Colors` in it's own group - Color Bar - under Tree view. I ended up with a default text size of 13 but that's an easy change if you think it's too small/large. I'm just waiting on the CI to give me a green tick and one final test here and then you'll be able to build it from a new branch I've not created yet which will have a single commit, will update this evening.

Syres916, 2023-09-02:
OK, @FEA-eng if you want to compile and test https://github.com/FreeCAD/FreeCAD/compare/master...Syres916:FreeCAD:Fix_ColorBar_Position_and_Text and let me know if there's anything that needs tweaking.

FEA-eng, 2023-09-04:
@Syres916 Thank you. Unfortunately, currently, I'm unable to correctly compile FreeCAD from the source (it's compiled in such a way that I can't create any geometry). So I would appreciate anyone helping us test this. Meanwhile, I'll try to resolve those issues but it might be difficult.

luzpaz, 2023-09-04:
I may be able to create a linux Snap package to test this on...any interest?

FEA-eng, 2023-09-04:
@luzpaz So far, I've been using Windows and trying to compile on this system but I guess that it's about time for me to set up a Linux environment to be able to help more with FreeCAD development.

FEA-eng, 2023-09-06:
@luzpaz and @Syres916 Now at least I have access to virtual machine with Ubuntu so I could probably test it using that snap package (I will have to learn how to use it, I’ve only used AppImage so far).

FEA-eng, 2023-09-06:
Hg

FEA-eng, 2023-09-06:
I closed it accidentally

Syres916, 2023-09-15:
@FEA-eng @luzpaz I've requested another maintainer to give the once over the PR but I am loosing in faith in the whole process. There's been not one piece of real feedback negative or positive regarding the actual code. The longer it goes on the more chance of significant conflicts that then have to be resolved.

FEA-eng, 2023-09-15:
> @FEA-eng @luzpaz I've requested another maintainer to give the once over the PR but I am loosing in faith in the whole process. There's been not one piece of real feedback negative or positive [...]

I can at least try to test this. I'll try to build FreeCAD on Ubuntu virtual machine. If it fails again, I'll consider manually modifying the files and triggering AppImage like I did with #9477. The lack of maintainers for the FEM workbench is indeed a huge problem but we may still convince other devs like Chris to merge this if we confirm that it's working as expected. Recently, our changes to naming and availability of non-working features in the FEM workbench were merged this way. I hope that not-so-big and tested PRs can still be handled like this despite the lack of mergers for this workbench.

luzpaz, 2023-09-16:
I opened up a separate issue to make an official call for maintainership (interim or long-term) :point_up_2:

closed 2023-09-19 by chennes