FreeCAD/FreeCAD #10439: [Problem] Labels of the FEM legend are poorly visible

FEA-eng, 2023-08-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33992 (Git)
Build type: Release
Branch: master
Hash: 90bafb3a069468455d8c65d030ef8750b510c39b
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
[/code]
```

### Subproject(s) affected?

FEM

### Problem description

Labels of the FEM legend are too small (no option to increase their size or bold them) and poorly visible due to their black color (no option to change that either - see #9617).

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-08-31:
Getting somewhere 
[image]

Just need to decide the best place for the UI change, I was going to add it to this task panel:

[image]

Any thoughts?

FEA-eng, 2023-08-31:
@Syres916 Thank you for looking into this. It would be best to add the label color and font/size options to that context menu in your screenshot but I wouldn't mind it being in the preferences for the FEM workbench as well.

Syres916, 2023-08-31:
The reason for not having it specifically for FEM is the colorbar is used elsewhere in FC, the screenshot above is from Mesh-CurvatureInfo which is where I've been doing my testing.

closed 2023-09-19 by chennes