FreeCAD/FreeCAD #10441: [Problem] The Mesh group and Node set tools in FEM are most likely broken

FEA-eng, 2023-08-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33992 (Git)
Build type: Release
Branch: master
Hash: 90bafb3a069468455d8c65d030ef8750b510c39b
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
[/code]
```

### Subproject(s) affected?

FEM

### Problem description

I was testing the Mesh group and Node set tools in the FEM workbench. It seems that they are not working properly or are extremely unintuitive (less likely). Moreover, they are very poorly documented - the Mesh group tool has a wiki page (https://wiki.freecad.org/FEM_MeshGroup) but omits important information regarding what to select (apparently, the tool expects geometrical entity selection), what export formats are supported and so on. The Node set tool has no description at all: https://wiki.freecad.org/FEM_CreateNodesSet

Both tools seem to be working at first (apart from several options in the Node set tool like Pick instead of Poly) but have no effect on the exported mesh or generated CalculiX input file.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

closed 2024-02-26 by chennes

ceanwang, 2024-12-20:
It works perfectly.  See https://forum.freecad.org/viewtopic.php?t=93086

Strongly suggested this be reversed back.

FEA-eng, 2024-12-20:
> It works perfectly. See https://forum.freecad.org/viewtopic.php?t=93086
> 
> Strongly suggested this be reversed back.

IMO it's far from perfection. Some selection modes are indeed working (Pick isn't, right?) but what can you do with a node set created this way? Apparently, only access it from Python console. This tool was supposed to create node sets for export and/or use with other constraints.

To bring it back, it would definitely need some changes. At least removing non-functional options but I think that it needs more usability (and possibly user-friendliness).