FreeCAD/FreeCAD #10442: [Feature Request] Symmetric option for Linear and Polar patterns

kadet1090, 2023-08-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
OS: Arch Linux (awesome)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: makepkg
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.11.3, Qt 5.15.10, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly3 0.12.0
  * IconThemes
  * Glass (Disabled)
  * lattice2 1.0.0
  * A2plus 0.4.60n
```

### Subproject(s) affected?

PartDesign

### Problem description

For some patterns it'd be nice to be able make them symmetrical to the original feature(s). I think that the best option would be to implement it as additional boolean property / checkbox for the patterns - this way we could achieve similar UX between two possible pattern kinds. 

### Open questions

#### How it should behave?
There are few options how such "Symmetric" option could work, for example, when we have:
```
- Angle set to 90
- Mode set to Overall Angle
- Occurrences set to 2
- Symmetric option on
```

Should we have:
a) 2 features, 90 degrees apart? We would have to "move" original feature.
b) 3 features spanning across 90 degrees?
c) 3 features spanning across 180 degrees - 2 per symmetry side, each side spans on 90 degrees? Then we'd have to constraint maximum angle to 180, not mentioning that angle would no longer be overall.

[image]

Personally I think that a) seems to be most useful, c) also could be an option but it could be easily done now using multi transform with pattern + mirror. I however think that having that option as simple checkbox would be great for product usability. 

If we choose option a) for even number of occurrences, I see few potential problems:

#### Relation to Topological Naming
As there will no longer be feature in original place, instead pattern would be distributed evenly across length or angle. Will it have impact on topological naming, as it'd be hard to name any created feature as original?

#### Non-Patterned feature
We can create pattern only for some features - how about case where we have some dependency on the patterned feature which will not be a part of pattern? Should we even care, or consider it as error from the users side which should be corrected (feature should be created after pattern).

[image]

For this example the pocketed circle will move so peg will no longer be placed correctly.

### Anything else?

Forum discussion: https://forum.freecad.org/viewtopic.php?t=68621&start=13

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mbello, 2023-08-31:
I always missed this feature. But what if, instead of "Symmetric", which has all the corner cases you mentioned, we had a simpler "Mirrored" boolean?

grubertechnic, 2023-08-31:
There is already a mirror feature for patterns. But after rethinking the cases, a symmetric makes sense, especially when more options of transformation will be added. But it makes only sense IN the pattern itself.

> But what if, instead of "Symmetric", which has all the corner cases you mentioned, we had a simpler "Mirrored" boolean?

That is the point, we already have mirror, for all cases there will be needed a symmetric.

kadet1090, 2023-08-31:
Mirrored would indeed be much simpler, but it's not edge-cases free - see case with polar pattern for angles above 180, especially 360 where it'd create doubled features. If we won't be able to converge on behavior for symmetric case I'd personally still like to see "Mirrored" checkbox. At least it'd be easier to create such patterns and require less operations from the user - which always is better from UX perspective.

adrianinsaval, 2024-01-23:
I'm not convinced that we need or want this feature, we have multi transforms where you can make a mirror out of a pattern. This is duplicating functionality in a different place.

obelisk79, 2024-01-23:
I agree, duplicating functionality is generally a bad idea and is even addressed in the design guide if I'm remembering right. Perhaps @kadet1090 can explain how this is different or perhaps is a superior approach?

kadet1090, 2024-01-24:
Yes, implemented as Mirrored it is duplicating thing that we already functionally have, however making it easier to discover. Multi transform works but the workflow is not easy to find, not obvious - and it is still mentally two steps to define. 

That said, this issue was mostly about symmetry so options a and b which are hard to define correctly. I even stated in the description that c is already easy to achieve and thus not as useful as for example option A which allows to easily distribute feature knowing the symmetry and overall length which now must be done using expressions.

Honestly, I don't think that we agreed on how it should work here and that is why I never bothered to implement it.

adrianinsaval, 2024-01-25:
I agree multi transform has bad UX, but the solution is to improve the UX there not to hack duplicated functionality somewhere else.

adrianinsaval, 2024-01-25:
On the other hand Pad and revolve offer a symmetry option, so it might be justified after all.

maxwxyz, 2026-11-07:
Part of this is in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/7d506c94f95cb641969fa2d84a2931d8e970159c

Linear and polar patterns can mirror the copies to both sides. The count is per side including the original, so this gives an odd total, it does not cover the even-count variant discussed here.