FreeCAD/FreeCAD #10443: [Problem] Sketcher_Dimension doest not allow selection->tool workflow

PaddleStroke, 2023-08-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
0.22 latest
```

### Subproject(s) affected?

Sketcher

### Problem description

I noticed that the newly introduced tool Sketcher_Dimension does not allow the workflow where you select the geos first, then starts the tool.

There's no reason why it shouldn't support this so I think it should be implemented.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

howie-j, 2023-08-30:
Yes, i also noticed this. I personally do not use that workflow, but that is how much of Freecad works so it should probably be implemented.

luzpaz, 2023-09-04:
@PaddleStroke please use `Fixes #10443` next time to auto-close tickets when their PRs get merged, TIA!

closed 2023-09-04 by luzpaz

PaddleStroke, 2023-09-04:
Thanks I didn't knew !

luzpaz, 2023-09-04:
@PaddleStroke :+1:  
JFYI, see https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue