FreeCAD/FreeCAD #10453: [Problem] Sketch editor freezes

rabirland, 2023-08-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Windows 11 build 22000
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Curves 0.6.13
  * freecad.gears 1.0.0
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

The sketch editor interface frequently freezes during editing a sketch. It only happen when a specific user action happens, this includes:

- Hovering a feature / constraint
- Scrolling
- Zooming

It only and ever happen when part of the sketch is visible on the screen. No user action causes freeze when nothing is visible on the screen from the sketch (not even hovering the axis lines). In 0.20 the issue arose in completely random occasions, since upgrading to 0.21 it never occurred when editing a fresh document, but is more frequent after opening an existing project. Altering the rendering / caching options of the sketcher had no effect.

One way to mitigate the issue is to zoom in on the sketch as much as possible, so that only a very tiny amount of features visible at once, or edit the constraints from the properties list without opening the sketch to edit, but it is not ensured the freeze won't happen.

Recomputing, editing sketch (outside the sketch editor) or using the sketch in any other feature had no effect, it only happens inside the sketch editor.

Opening the sketch for edit won't freeze the application **until** a user action inside the sketch editor window. The other parts of the UI (windows, toolbars, status bar, etc...) won't trigger a freezing.

Here is an example, zooming on the sketch beforehand, then slowly scrolling to move the sketch into the visible area. The app won't freeze until part of the sketch visible. Even then, it didn't freeze while only a few number of feature was visible.
https://youtu.be/LYqUVuM1D9c

Here is a video without zooming in. Opening the sketch for edit didn't freeze the app, but the moment I tried to scroll, it did (you can see it did scroll a little bit before freezing):
https://youtu.be/PgEeIXm6MGw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rabirland, 2023-08-31:
> Does it mean, in 0.21 it is solved?

No, the videos was recorded in 0.21, it's just not that predictable in 0.21 as it was in 0.20. It was a fresh install (after a system reinstall) and the issue still exists. I checked my sketcher, the auto-upate is turned off, so that was not the issue, also it shouldn't recompute when you simply hover over a item, or should do the same when heavily zoomed in too, but in reality the zoom level seem to affect whether it freezes or not.

rabirland, 2023-08-31:
Found a repro step in 0.21:

- Create a new document,
- Create a body
- Create a sketch (any plane)
- Add a hexagon
- Tangent the outer circle to the two axes
- Constraint the circle with a diameter
- Contraint a face of the hexagon with a vertical/horizontal constraint
- Pad the sketch (any length)
- Save file
- Close FreeCAD
- Reopen freecad, open the previous file
- Open the sketch
- Start to zoom out, at one point it freezes.

Video of the sketch: https://www.youtube.com/watch?v=tN4Yfrp2qYQ

rabirland, 2023-08-31:
Update: I think I found the reason, it's the `Display => 3D View => Render Cache`

Having it on "Auto" or "Distributed" will result in a freeze, but it looks like setting it to "Centralized" fixes the issue.

grubertechnic, 2023-08-31:
Thanks for the very detailed description!

I really followed you steps 1:1, but still can not replicate it. Also when i open your file, everything is fine, i can zoom in and out, pan around, everything. But also i am using 

```
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34095 (Git)
Build type: Release
Branch: master
Hash: f3463fe5a7fe1ca00bbd94edfa643814db114e65
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
```

mean,s my version is newer.

You can try the newest weekly build, see https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds.

But when, please rename the FreeCAD config folder `%APPDATA%\FreeCAD` first, then test it, (configs will be different), and after testing, delete the new created folder for testing and rename back the old config file. Reason: when open the old config file with 0.22, it could be, that some thing in the config changed, for 0.22 no issue at all. But when opening the 0.22 config file with 0.21, unexpected things can happen.

Or we wait for someone, who can already test it (much saver).

rabirland, 2023-08-31:
Tested with the newest weekly, with the old config removed, still the same:
https://www.youtube.com/watch?v=kQJD9p_rCnM

grubertechnic, 2023-08-31:
Yes i see. For me, pure guessing, a graphic card/driver issue. As i said, we had in the past rarely some sketcher graphic card/driver issues. We must at least wait for someone, who can replicate it.

rabirland, 2023-09-22:
Update, now using the 0.22 version
```
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33992 (Git)
Build type: Release
Branch: master
Hash: 90bafb3a069468455d8c65d030ef8750b510c39b
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C) [ OS: English/United States (en_US) ]
Installed mods: 
  * Assembly4 0.50.2
  * Curves 0.6.13
  * fasteners 0.4.63
  * freecad.gears 1.0.0
[/code]
```

It seem like, the freezing gets "progressively worse" over time, regardless of the installed version. But going into the Preferences => Display and changing the Anti-aliasing to None and Render Cache to Centralized, saving and restarting the app "resets" the freezing (even if these settings are changed back to the original values.

Manually clearing the cache has no effect.

craysiii, 2023-11-21:
I am running into this issue as well. Opening a previously saved file and opening a sketch on a part freezes FreeCAD after attempting any sort of view operation in the window.

[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Curves 0.6.12
  * pcb 6.2023.0
[/code]

I can provide a file if needed, but it has happened to three separate files I've tried to open. This does not happen on new documents, only existing ones which have been opened in a new instance of FreeCAD.

luzpaz, 2023-11-26:
Is this perhaps a Qt issue ?

rasgit23cb16, 2023-11-29:
I have the same issue, i just updated my FreeCAD from 0.20 to 0.21, when i open a file and just move (with right mouse button - input mode gesture) it freezes within 0.5 secs.
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
[/code]

voidexp, 2024-02-10:
FreeCAD 0.21.2, 100% sketcher freeze reproducibility with this simple project: https://1drv.ms/u/s!AtE-O7epzHPGga5xem9s7Sb1152EJA?e=grF0Ut

1) open the project
2) open the Revolution->Sketch for editing
3) hover over the center of the sketch and experience complete freeze

I've noted that if I open the sketch for editing with double-click and DO NOT enter the 3D viewport area with the cursor, but for example hover the mouse over toolbars or the document tree, all is good.

xtemp09, 2024-02-10:
voidexp, I can't reproduce.

```ruby
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

progserega, 2024-05-26:
[freecad.strace.gz](https://github.com/FreeCAD/FreeCAD/files/15445989/freecad.strace.gz)

In my project all skech freezes. I get strace. 
Linux debian 12
freecad 20.2
amd gpu
amd64

xtemp09, 2024-05-26:
@progserega, could you share the example file, the reproduction steps, Help → About info? Perhaps, _FreeCAD.log_ would be good, you can turn logging on with `-l` command-line option.

maxwxyz, 2024-07-01:
@chennes please test on a Qt6 build

benj5378, 2024-09-02:
How does CPU look on freeze? 100% or 0%?
If run in debugger and Ctrl + C, is there any interesting back trace?
If run in callgrind (analyze with KCachegrind), are there any interesting functions taking up a lot of time? Make sure to spend majory of time in freeze mode

kachurovskiy, 2024-09-16:
I might have hit the same issue with https://www.reddit.com/r/FreeCAD/comments/1fhmxuy/monstrously_slow_on_simple_edits/ - there's a link to a file there and several people reported they reproduce.

luzpaz, 2024-10-23:
@kachurovskiy the link to file is obsolete. Can you zip it up and drag/drop it into github textbox of this ticket ?

closed 2025-06-23 by github-actions[bot]