FreeCAD/FreeCAD #10454: [Feature] Add an icon to create reference constraint contextually

FlachyJoe, 2023-08-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Debian GNU/Linux trixie/sid (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33981 (Git)
Build type: Unknown
Branch: master
Hash: 5a153e50ffb6046b6cc4dba498344032f6edb952
Python 3.11.4, Qt 5.15.10, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: French/France (fr_FR)
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

 #9810 introduce a unique tool to create a dimensional constraint. The workflow to create a reference one is now

1. click Toggle driving/reference to activate reference mode
2. click Dimension, do the needed stuff
3. click Toggle driving/reference to go back to driving

A Reference Dimension button can reduce the workflow to one click.
So I suggest to add a ReferenceDimension icon next to the Dimension.

### Anything else?

Ping @PaddleStroke 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2023-08-30:
Sketcher_ToggleReferenceConstraint is still needed to switch an already created driving constraint to reference or back.

PaddleStroke, 2023-08-31:
Sketcher_ToggleReferenceConstraint could be removed as one can switch an already created constraint by double clicking it. I don't say it should be removed, just that it could.

obelisk79, 2023-08-31:
I don't think this is a great idea. I'm all for efficiency, but adding more icons to the newly simplified toolbar seems like taking a step back, not forward. Couldn't this just as easily be toggled by the keyboard instead?

PaddleStroke, 2023-08-31:
You can already toggle by the key shortcut I think. But we can't have no more visible UI way of switching between reference and not reference.
When tool settings merge perhaps we could have a checkbox 'Reference (K,X)' in the taskbox instead of a toolbar tool.

howie-j, 2023-08-31:
I also dislike this idea.

Driving/Reference can be toggled already multiple ways:
- toolbar button
- shortcut key (default K,X)
- during constraint creation, check "reference" box
- after creation by selecting and modifying by all the above means

That means there is currently six ways of creating reference constraints. Adding another toolbar icon just for this seems unneccessary imo. Also there are not a dual set of create geometry commands, so this would create even less ux consistency.

maxwxyz, 2024-08-30:
Maybe the dimension tool could make use of the task panel widget as well which would also make it obvious that it is still active together with https://github.com/FreeCAD/FreeCAD/issues/12465 . There could be a checkbox for reference/driven constraints (default false).