FreeCAD/FreeCAD #10455: [Problem] Sketcher Dimension alternate constraints are sub-optimal

howie-j, 2023-08-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Fedora Linux 38 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34085 (Git)
Build type: Release
Branch: master
Hash: 4fb7b912b444005d2fa9745d96e0f70f81dac804
Python 3.11.4, Qt 5.15.10, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

Using _Sketcher_Dimension_ and selecting multiple points, when scrolling through available constraints using SHIFT the coincident constraint is not an option.

Also when selecting two lines, only _angle_ and _equal_ are available, not _parallel_.

IMO not all constraints must be available with the Dimension tool, however supporting the most used constraints as alternatives would make Dimension very powerful.

Suggestion for available constraints:
- When only points are selected: Coincident, horizontal, vertical, symmetric (3 points only)
- When only lines are selected: equal and parallel

Please discuss below.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

howie-j, 2023-09-07:
While the problem is written as if there is not enough alternate constraints available, that was not the intent. I was hoping for a discussion on which alternate constraints should be available depending on selection, and which, if any, should be removed.

For context, there has been quite a lot of opinions on this in the forum: [](https://forum.freecad.org/viewtopic.php?t=65521)

Some argue that since most geometrical constraints are "hidden" behind a modifier key they should not confuse new or inexperienced users, while other think Sketcher_Dimension should only create dimensional/non-geometrical constraints.

Please voice your opinion on the alternate modes of the Sketcher_Dimension, and lets make this tool even better, no matter if it becomes simpler or more advanced in the end :)

cuedcad, 2023-09-07:
See comments at [](https://forum.freecad.org/viewtopic.php?t=65521&start=240#p704550)

maxwxyz, 2024-01-04:
@howie-j this would be possible. Is the improved context menu with right click a first option for your problem? #11884