FreeCAD/FreeCAD #10460: [Problem] App::PropertyPythonObject is not saving data

Serince, 2023-08-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.11.4, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * aa.qrc
  * FEMbyGEN 1.5.0
  * Plot 2022.4.17
  * Ship 2022.4.11
  * CfdOF 1.19.4
[/code]
```

### Subproject(s) affected?

None

### Problem description

I use App::PropertyPythonObject to store some data. When I save the file and open it again, no data comes in 0.21 version. When I try the same process in version 0.20.2, it works fine.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-01:
@Serince can you demonstrate the bug ?

Serince, 2023-09-01:
Sorry, I checked several times before I posted the error and there was indeed such a bug. Now when I try it on another computer to show it I don't have this problem, I think there is something wrong with my configuration. Let me reinstall freecad. I apologize for opening an issue without investigating the bug in more detail.

Serince, 2023-09-01:
Hello,

I catch the error again. The problem comes with python 3.11 . I have installed two conda environment to test it. The data is storing freecad 0.21.1 Python 3.10. But freecad 0.21.1 Python 3.11 has the bug. Not storing data in object as ProperyPythonObject. I add a small macro to store data. After running it, save the file and close / reopen the file. When you call the stored data it will return None.

import FreeCAD as App
doc=App.newDocument()
obj=doc.addObject("Part::Box","Box")
doc.recompute()
obj.addProperty("App::PropertyPythonObject", "Dictionary")
obj.Dictionary={"Stored data": [3,5,7]}
print(obj.Dictionary)

[image]
[image]

Syres916, 2023-09-02:
Confirmed using Python 3.8.10 and Python 3.11.2

```
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34020 (Git)
Build type: Release
Branch: master
Hash: 6200150fac0e311cefb4c4a2c3122455d2856251
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
```

Syres916, 2023-09-02:
If it's of use to the devs, extracting the Document.xml from each file created exactly the same way gives:

Python 3.11

```
<Property name="Dictionary" type="App::PropertyPythonObject" group="" doc="" attr="0" ro="0" hide="0" status="2097152">
<Python value="bnVsbA==" encoded="yes" json="yes"/>
</Property>
```

Python 3.8

```
<Property name="Dictionary" type="App::PropertyPythonObject" group="" doc="" attr="0" ro="0" hide="0" status="2097152">
<Python value="eyJTdG9yZWQgZGF0YSI6IFszLCA1LCA3XX0=" encoded="yes" json="yes"/>
</Property>
```

BTW, `value="bnVsbA=="` translates to `Null`

I checked the json library version on both and they are the same version: 2.0.9. Could this be yet another encoding issue??

Syres916, 2023-09-05:
If the PR is accepted then I'll raise a backport PR so 0.21.2 has the fix.

wwmayer, 2023-09-06:
File created with FreeCAD using Py3.10:
[issue_10460_Py310.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/12539795/issue_10460_Py310.FCStd.zip)

closed 2023-09-24 by chennes

s-quirin, 2024-01-17:
I'd like to point out that people have to implement the methods `dumps` and `loads` to avoid errors since the release of FreeCAD 0.21.2 and keep `__getstate__` and `__setstate__` (for FreeCAD < 0.21.2). Was this intended for this minor realease?

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * FEMbyGEN 2.0.0
```

@Serince: issue addressed in PR

Syres916, 2024-01-17:
@s-quirin add-on PRs are being dealt with piecemeal, see https://forum.freecad.org/viewtopic.php?p=729512#p729512
There was a minor backport that didn't make it into 0.21.2 but will be in 0.21.3

commit 83d4080f (2023-09-24): App: fixes #10460: App::PropertyPythonObject is not saving data

Since Py3.11 the methods names __setstate__ and __getstate__ conflict with the method names added to the object class.
Thus rename them to 'loads' and 'dumps'
commit fbe2fef6 (2023-11-10): App: fixes #10460: App::PropertyPythonObject is not saving data

Since Py3.11 the methods names __setstate__ and __getstate__ conflict with the method names added to the object class.
Thus rename them to 'loads' and 'dumps'