FreeCAD/FreeCAD #10461: [Problem] PartDesign | Boolean : Boolean body displayed in "Tools" mode after restart

CopperWorm, 2023-08-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
```

### Subproject(s) affected?

None

### Problem description

PartDesign | Boolean : Boolean body displayed in "Tools" mode after restart

After reload of file with boolean body stored with display mode set to "Result" it comes up with display mode set to "Tools".

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-04:
@CopperWorm can you create a screencast showing the issue ? Maybe also uploading an example file (zipped)

CopperWorm, 2023-09-11:
Hi luzpaz, 
attached, you'll find the example file.
After loading you'll see the boolean in tools mode. Please switch the boolean view to result mode, then save it and reload it... 

Thank you guys for your work and dedication !
Michael

[boolean_body_displayed_in_tools_mode_after_reload_or_restart.zip](https://github.com/FreeCAD/FreeCAD/files/12577548/boolean_body_displayed_in_tools_mode_after_reload_or_restart.zip)

luzpaz, 2023-09-30:
@macdroid53 do you have a moment to triage this ?

macdroid53, 2025-01-21:
Seems ok on dev version 39861.

closed 2025-06-21 by github-actions[bot]