FreeCAD/FreeCAD #10467: [Problem]  Problem: User can't indicate what type of constraint to add

sliptonic, 2023-08-31:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Full version info

N/A

### Subproject(s) affected?

None

### Problem description

A fundamental feature of the assembly workbench is for the user to constrain how one part moves relative to another.  Numerous different types of constraints are possible and it should be efficient for the user to indicate exactly what type of constraint to apply based on what geometry is selected and what else the user chooses.  For example, the user may select a single face from each of two solids. They may wish to make the two faces co-planar or simple parallel.

The user needs to be able to select geometry from two parts and apply an appropriate constraint between them.

This issue will start the discussion about how the user workflow will function for applying constraints.

### Anything else?

[constraint grid](https://help.spaceclaim.com/2015.0.0/en/Content/Printable/SpaceClaimAssemblyConstraintsChart.pdf)

[constraint type overview](https://help.alibre.com/articles/#!alibre-help-v27/assemblies-assembly-constraints-types-of-assembly-constraints-overview)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-09-01:
There are different approach to this problem.
- Solidworks approach is similar to A3 and A2+ : You select features of the part (faces, edges, points) then you choose what kind of constraint you want to apply: parallel, distance, co-axial...
- OnShape/Fusion approach are using 'mate connectors' (a bit similar to A4) a short video will be more efficient than words : https://www.onshape.com/en/resource-center/videos/build-your-first-assembly

Maybe the first question is which workflow do we want to achieve ?

tobiasfalk, 2023-09-01:
Maybe a lost of benefits and drawbacks of each method would be good. Also is one method is preventing future features, like that two parts of a assembly cam be moved(in the programm) or that for example a FEM simulation is easier with one?

sliptonic, 2023-09-01:
Another good reference on constraint types.
https://forum.freecad.org/viewtopic.php?t=80819

FEA-eng, 2023-09-01:
> or that for example a FEM simulation is easier with one?

That won't be a problem. FEM completely ignores assembly constraints and a user has to define all mechanical constraints (also operating on degrees of freedom but in a more direct way) to be used in the analysis in the FEM workbench.

tobiasfalk, 2023-09-01:
@FEA-eng 
>That won't be a problem. FEM completely ignores assembly constraints and a user has to define all mechanical constraints (also operating on degrees of freedom but in a more direct way) to be used in the analysis in the FEM workbench.

But if a Constraint System like the one @sliptonic linked to is used, wouldn't that allow FreeCAD to find the degrees of freedom by its self?

>Another good reference on constraint types.
https://forum.freecad.org/viewtopic.php?t=80819

Also, this was just what came to my mind for examples where one System could interfere with future features.

tobiasfalk, 2023-09-01:
Just a start of a List of benefits and drawbacks, 

# Constraint System
@PaddleStroke 
>Solidworks approach is similar to A3 and A2+ : You select features of the part (faces, edges, points) then you choose what kind of constraint you want to apply: parallel, distance, co-axial...

@sliptonic 
>Another good reference on constraint types.
https://forum.freecad.org/viewtopic.php?t=80819

This system locates parts by constrains between features of those parts.

## User
### Benefits

- Efficient and precises in connecting two parts.
- ...

### Drawbacks

- Hard to learn, because there are multiple commands or ways two parts can go together, this can be mitigated with a good Tutorial and Documentation (maybe even a good video tutorial showcasing each constraint)
- ...

## Coding
### Benefits

- ...
- ...

### Drawbacks

- ...
- ...

## Future WB Features
### Benefits

- ...
- ...

### Drawbacks

- ...
- ...

# Mate Connector System
@PaddleStroke 
>OnShape/Fusion approach are using 'mate connectors' (a bit similar to A4) a short video will be more efficient than words : https://www.onshape.com/en/resource-center/videos/build-your-first-assembly

This system locates parts by mate connectors, which are part of a Part, and one connector connects to the mate connector of another part. 

## User
### Benefits

- ...
- ...

### Drawbacks

- ...
- ...

## Coding
### Benefits

- ...
- ...

### Drawbacks

- ...
- ...

## Future WB Features
### Benefits

- ...
- ...

### Drawbacks

- ...
- ...

# How to use/expand
Copy and expand this when you have a Benefit or Drawback for one of the systems and categories.

FEA-eng, 2023-09-02:
> But if a Constraint System like the one @sliptonic linked to is used, wouldn't that allow FreeCAD to find the degrees of freedom by its self?

I think that the constraint concept in FEM is so different than in assemblies that it wouldn't make sense to convert between assembly and FEM constraints. This level of automation could also lead to many mistakes.

To show you how different it is - in FEM you typically select a geometric entity (solid, face, edge or vertex) and specify which DOFs (usually in a global coordinate system) should be fixed or enforced (e.g. you want a face to move by 2 mm). Internally, those boundary conditions are applied to nodes associated with that geometric entity. That's for entities not connected to others. Assemblies typically don't have such constraints, one would have to use lots of datum planes and axes to achieve the same effect. Then there are cases when geometric entities are connected to others (e.g. 2 touching faces). Then one can glue them using tie constraint (no relative motion possible) or apply contact which is basically a form of collision detection (but much more complex since it involves the calculation of transmitted forces, often with friction).

maxwxyz, 2023-09-02:
Would it be possible that the WB only present valid constraints which make sense, based on the features the user has selected before clicking a "constraint icon"? Maybe it is also possible to show the result when hovering over different constraint types in the menu / button with a animation (show the remaining DoF). Or maybe after applying the constraints, a "check DoF" button which animates/wiggles all parts in their non-constraint DoFs.

sliptonic, 2023-09-02:
> Would it be possible that the WB only present valid constraints which make sense, based on the features the user has selected before clicking a "constraint icon"? 

Maybe.  This kind of 'contextual constraint' makes a lot of sense.  In the 'constraint grid' link in the first post is the way spaceclaim views it.  In their model (which I'm not advocating) there would only be three buttons on the toolbar: 'tangent', 'align' and 'orient'.   The specific constraint applied would depend on the selected geometry and which button is pushed.  Other assemblers that I've looked at have more buttons but do something similar.  

> Maybe it is also possible to show the result when hovering over different constraint types in the menu / button with a animation (show the remaining DoF). Or maybe after applying the constraints, a "check DoF" button which animates/wiggles all parts in their non-constraint DoFs. 

These are good feature ideas but probably in the category of UI improvements after 1.0 as opposed to essential parts of the workflow.  Eliminating redundant constraints or identifying over-constraint is probably an essential feature.

> If a user chose the "wrong" constraint, is it possible to change the constraint type in the model tree (while keeping all selected references) or does the user have to delete the wrong constraint and create a new one?

QtCreator has a 'morph' feature to turn one thing into another.   I could imagine something similar.  It's another UI feature that would probably be easy to add later.  For 1.0, if the user has to delete and recreate the constraint, it isn't the end of the world.

tobiasfalk, 2023-09-02:
Question, why a button for every constraint? Why not a "create constraint" Button, where the user then selects two parts/part-fetures and a list of possible constraints is provided. This would streamline the process of creating a constraint and the Button panel would not be clouded with a huge number of button that all create a constraint. But this is an Ui design question and not an underling system question.

tobiasfalk, 2023-09-02:
@FEA-eng I think you may have misunderstood me, if I am right the current FEM workbench only can compute one part simulations and not multipart simulations. But with a multipart simulation I would need to define how two parts are connected to each other and for this a Mate Connector System would be of not much help, or not?
This lies on the base thought that the Constraint System defines how two parts are touching/working with each others. The sliding pivot constraint, for example, defines that it cans travel and turn in one axis (relative between those two parts) but not in others.
In a multipart simulation, there is the DOF between two parts and the DOF of the assembly (multipart object). 
Yes, a multipart simulation would require the inclusion of many more parameters like friction and so.

So is the creation of a multipart simulation feature for FEM based on an assembly with the Constraint System easier than with a Mate Connector System?

As for why I want the conformation of someone else, I have no notable experience in FEM simulations. Especially if it comes to coding for it.

grubertechnic, 2023-09-02:
A FEM constraint have nothing to do with an assembly constraint, you can not mix it.

Minimum example: a simple bearing, even it does not work the same as constraints in its simplest case. And to be explicit, a roller bearing (if you can simplify it with FEM btw.)  have a different FEM constraint (one sided pressure (better sectional) or force) as a glide bearing. And to be more explicit glide hydrostatic glide bearing have a extreme different FEM constraint (one sided pressure and/or force) against to a hydrodynamic bearing (around pressure and addition sided pressure or force).

In assembly a bearing is a simple axial constraint (and when it is the thrust/fixed bearing in addition an additional constraint)

PaddleStroke, 2023-09-05:
After a discussion with the author of the Ondsel Solver that is going to be solving assemblies, it seems that this solver use 'joint' concept very close to the way OnShape is doing things: fix joints, revolute joints, slider joint ...
So I suggest that we implement this workflow as a starting point.

I will start implementing a UI around this workflow.

jpg87Fr, 2023-09-14:
I don't know how to program, so it's inappropriate for me to make requests, but give opinions, why not?
Indeed, I find it a shame to always “reinvent the wheel”.

There are currently several assembly WB, each with its supporters, its advantages, and perhaps its disadvantages.
Personally, I am a fan of Assembly3 WB, but I understand very well that other users have other preferences, for example the approach of Assembly4 WB which is very different.

So rather than creating yet another new WB, wouldn't it be more useful to ensure that the particularities of each current WB can coexist without disturbing each other, allowing each user to work as they wish, while having files readable in all cases?

If I understood correctly, you need an Asm3 assembly container which has its own characteristics to contain the “constraint elements” on the one hand, and the assembly constraints on the other hand.
Asm4 needs to associate an LCS with its components and probably other things that I'm forgetting since I don't work with this WB.
Can't we create a single container that accommodates the characteristics of each WB?

I saw that you are relying on Spaceclaim paintings to build the constraints of your new WB.
Have you paid careful attention to the constraints that Asm3 WB offers? I know, the abundance of constraints is scary. But below are the only constraints that are really useful for working efficiently. No need to complicate your life. The profusion of constraints in Asm3 WB comes from what was offered for its Solvespace solver.

[image]
[image]

Where is the problem ? Why aren’t you looking to work with @Realthunder? Didn’t he make his source codes free? Wouldn't it be more reasonable to only work on a solver whose license would be compatible with that of FreeCAD to resolve the problem of its distribution?

FEA-eng, 2023-09-14:
> So rather than creating yet another new WB, wouldn't it be more useful to ensure that the particularities of each current WB can coexist without disturbing each other, allowing each user to work as they wish, while having files readable in all cases?

There's really no point in questioning the decision to create a built-in assembly workbench at this point. It has already been discussed extensively in the past and the development is in progress. It might seem that creating an assembly workbench from scratch is a lot of unnecessary work but, judging by the PRs, it's not the case. The kinematic solver is ready, the interface is being worked on and the constraints are now almost ready too.

Also, the fact that a new built-in assembly workbench was introduced doesn't mean that the remaining add-on assembly workbenches will be somehow unified, making it possible to read their files even when they are not installed. This has also been under consideration. The problem is that those add-on workbenches are created by independent users and it's hard to convince them all to unify everything.

IMO it's also better that the new assembly workbench will have its own set of constraints instead of just copying them from some other assembly wb. This way, we will have one basic official assembly wb and a few add-on workbenches with different capabilities and workflows for users to choose from if needed.

jpg87Fr, 2023-09-14:
Something is very frustrating for users in the process: I tried to follow the different discussions on the forum. But the most important discussions have slipped to Github, where clumsy users (I understand from the discussions that I'm not the only one in this situation) have difficulty keeping up.

How can we get an idea of ​​the constraints that are currently being constructed (almost finished, you say)? 
How do you get an idea of ​​what’s coming when you’re not a programmer? 
As is often the case, users have little say in the matter. When the job is done, they have a choice: either they adapt or they will...

I don't want to create controversy, but why would independent programmers be less reliable than programmers from a private company? Seen from the outside, it may be suspected of directing its developments according to its own needs, even if this is not the case.

On the other hand, such an operation does not encourage the participation of those who are not part of the “hard core”. I thought FreeCAD came from cooperative work...

JohnOCFII, 2023-09-14:
> Something is very frustrating for users in the process: I tried to follow the different discussions on the forum. But the most important discussions have slipped to Github, where clumsy users (I understand from the discussions that I'm not the only one in this situation) have difficulty keeping up.

Have you seen the multi-part articles on the need for a new approach to Assembly in FreeCAD?  This, along with discussion at last year's FOSDEM as well as the recent FreeCAD Hackathon ( https://blog.freecad.org/2023/05/24/freecad-hackathon-11-13-august-2023/ ) seem to be the best way to stay up to date on the core developments in this area.

https://ondsel.com/blog/default-assembly-workbench-1/

PaddleStroke, 2023-09-14:
The goal here is to do a cooperative work and to build from the ground up a great assembly wb.
Could we have started by just merging A3 or A4 ? Yes probably. And this was initially the prefered solution. However there was many hurdles in this path.
- A4 has only fixed joints. No constraints at all. Its UI is needlessly complicated. On top of that its UI is inconsistent with the rest of freecad: no taskbox only dialogs. And code quality is not impressive.
- A3 UI is overly complex. Many tools shouldn't even be in assembly. RealThunder code is complex so he would have been the only one to own it.

After a deep review of all the assembly workflows around the place, it appeared that A3 is similar to solidworks but with a super-complex UI. And A4 is similar to onShape/Fusion but with a complex UI and only fixed joints.

After some discussion with Dr Koh, the maker of the open solver that we can use, it seems that the OnShape/Fusion workflow is the one that match the best. So I decided to offer to adopt this joining system. In the opened PR, the job is not 'almost done' as you mentioned.

Yes #10611 is a big step as it's defining the joining system, so it was hard to involve more people in its design. However it is far from done and if this is accepted it will open the way for a far more collaborative work as it will open the way to many things to do and improve.

kadet1090, 2023-09-14:
> [image] [image]

For me many of these constraints are simply redundant, and should be rethought to fell more intuitive.

For example we've got attachment and attachment offset. From the UX point of view it probably should be one constraint - attachment which could have property like offset applied. There is no need to separate those as it creates more complexity in the UI.

Same thing goes for Point constraints - it could easily be one constraint on the UI - point on object. You have selected point and line? Nice you have point on line. You have selected point and point? That'd be coincident. 

So it's not like there should be no changes. If freecad wants to be good and capable piece of software then it must be consistent and feel like one coherent tool and not few random pieces of code glued together. And merging any of existing assembly workbenches without a lot of rework would be simply gluing another random piece of code to freecad. So it is probably easier to write new solution based on experiences and code available in already developed modules.

sliptonic, 2023-09-14:
> For example we've got attachment and attachment offset. From the UX point of view it probably should be one constraint - attachment which could have property like offset applied. There is no need to separate those as it creates more complexity in the UI.

> Same thing goes for Point constraints - it could easily be one constraint on the UI - point on object. You have selected point and line? Nice you have point on line. You have selected point and point? That'd be coincident.

Exactly right

> And merging any of existing assembly workbenches without a lot of rework would be simply gluing another random piece of code to freecad. So it is probably easier to write new solution based on experiences and code available in already developed modules.

The various assembly options have their respective fans and they should remain viable options.  The goal of an integrated workbench isn't to kill off the others.  Quite the opposite.  So many lessons have been learned from all of them that we should be trying to stand on the shoulders of giants.  

Trying to merge one of the existing WBs sounds like an easier option but it isn't.  Not only would it be likely to alienate a huge part of the community, it would also be technically difficult for the reasons already expressed.  It makes sense to learn from all of them and build a solid foundation that is very "FreeCAD-ish" with testable clean code, and a clean separation between GUI and app logic.

jpg87Fr, 2023-09-15:
First of all, thank you for your answers and your explanations.
To @JohnOCII
Yes I followed these discussions, and I am trying to understand what is there at my level!

To @PaddleStroke
I share the opinion that Asm3 is complex in appearance, but as I already said, many functions seem superfluous to me. On the other hand, it works very well.

to @sliptonic
Your comments on Github are those of a confirmed user: I comply with them and will endeavor to adapt.
But how can I be sure I don't miss the information telling me I can start testing this new assembly WB ? I don't yet have the reflexes to use Github.

to @kadet1090
I'm not sure I share your views. Maybe I'll have the opportunity to come back to it.

kadet1090, 2023-09-15:
> The various assembly options have their respective fans and they should remain viable options. The goal of an integrated workbench isn't to kill off the others. Quite the opposite. So many lessons have been learned from all of them that we should be trying to stand on the shoulders of giants.

Of course! And nowhere I am suggesting that they should be killed of, only that in current state they are really great as downloadable extensions, where there is simply less expectation of coherent behavior. I even think that It'd be really great if current assembly workbenches could be done as extensions / additional tools for built-in solution. For example user could download "SolveSpace assembly workflow (A3)" or "LCS based assembly workflow (A4)" which would simply add more tools to built-in workbench.

ppemawm, 2023-09-15:
> * A4 has only fixed joints. No constraints at all....

It seems that what you are missing is that the Assembly4 LCS **is** the constraint that defaults to a static joint.  I think of it as a universal constraint or mating connector for which the attachment offset can be tailored using variables and expressions.  Consider a threaded constraint:  the angle of rotation of a nut is dependent on the position along its axis when assembled to a screw.  That requires one simple expression in the Property View.  Change the axial position and the nut will rotate the correct number of turns.  Same for a spring constraint, gears ,etc.  Name a constraint and there is an expression(s) for it.  

Much of parametric modelling in FreeCAD is driven by properties and expressions so the LCS assembly concept seemed natural to me as a longtime FreeCAD _only_ user since V0.13.  You combine that with the Attachment Engine for alignment of the LCS and you have a powerful, flexible, no frills, elegant assembler that is virtually immune to TNP.  IMHO, obviously.

What is lacking from Assembly4, of course, is a solver for the casual user that does not necessarily depend on Sketcher and a user interface that has the look and feel of the FreeCAD GUI, whatever that is.  I suppose it will inevitably evolve to look and behave like SolidWorks and OnShape probably do.

I do not look forward to a myriad of specialized buttons and tree objects that must be used to constrain a real world complex assembly such as a Harley motorcycle engine, an ultralight helicopter, the F16 main landing gear, etc.

closed 2023-09-21 by sliptonic