FreeCAD/FreeCAD #10468: [Problem] Deleting feature allocates dependent feature to the wrong location

benj5378, 2023-08-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34091 (Git) Snap 758
Build type: Release
Branch: master
Hash: a2735416e1189566329b327d99f115ced7071b86
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
```

### Subproject(s) affected?

Part

### Problem description

I have this object that I perform a mirror of, and it's located in a group. When I delete the mirror feature, I expect the object mirrored to keep it's location in the hierachy: inside the group.

This is not the case, after mirror feature is deleted, the object is relocated to the root of the hierarchy.

[Screencast from 2023-08-31 23-30-51.webm](https://github.com/FreeCAD/FreeCAD/assets/12400097/2e6ac550-2c2e-42f0-b32a-b30d1b3b2feb)

### Anything else?

Unsure if this bug applies to other than mirror feature.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

benj5378, 2023-08-31:
This also applies when trying to create an Array of an object

luzpaz, 2025-01-26:
@benj5378 can you reproduce? (if so can you add version you could reproduce on? TIA)

FEA-eng, 2025-06-08:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.42095 (Git)
Build date: 2025/06/06 16:36:48
Build type: Release
Branch: main
Hash: 9c017b0dbf538c2570df7c3316355cdcdc3163f5
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

FEA-eng, 2026-10-25:
I'm not sure how I reproduced it. The original object is still under Group after mirroring, but the mirror object is placed outside of it.

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 - Build 27.20.100.9664/Intel/Intel(R) UHD Graphics 620
```