FreeCAD/FreeCAD #10469: [Problem] Draft Move, wrong selection after task finished

benj5378, 2023-08-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34091 (Git) Snap 758
Build type: Release
Branch: master
Hash: a2735416e1189566329b327d99f115ced7071b86
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
```

### Subproject(s) affected?

Draft

### Problem description

I want to move object1. I start by selecting object1 in the hierachy, then click the Draft Move tool, I click a reference point and then a point on object2 as the destination. After the move have finished, the selected object is now that of the destination, object2's padding.

This is really inconvenient, because if I want to move object1 again, I now need to go through the hierachy, find the object and select it again.

It would be both convenient and expected, that after the Draft Move task have finished, the selection before the task started would be preserved.

[Screencast from 2023-04-13 17-59-53.webm](https://github.com/FreeCAD/FreeCAD/assets/12400097/b8a4a17f-970c-44b1-8f91-1664f81531fb)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-09-01:
Wrong video?

benj5378, 2023-09-02:
> Wrong video?

yes, updated with new vid

Roy-043, 2023-09-03:
Confirmed. AFAICT one of the edges of the target object has to have been highlighted for the problem to occur.

In my initial test with 2 Draft_Circles and only using center snap, the problem did not occur as I snapped to the center of the face of the target and moved there without its edge being highlighted.

Roy-043, 2023-12-28:
Related forum topic:
https://forum.freecad.org/viewtopic.php?t=75617

commit 3b3ed05d (2024-02-12): Draft: fix wrong selection after commands

Fixes #10469

The code of the `end_all_events()` function is by Chris Hennes (chennes).
See https://forum.freecadweb.org/viewtopic.php?p=656362#p656362.
closed 2024-02-12 by chennes