FreeCAD/FreeCAD #10470: [Problem] Draft Move, no feedback when trying to move an unmovable object

benj5378, 2023-08-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34091 (Git) Snap 758
Build type: Release
Branch: master
Hash: a2735416e1189566329b327d99f115ced7071b86
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
```

### Subproject(s) affected?

None

### Problem description

Newbie user here: hey, I want to move an object. I go select the Pad of my body, click the Draft Move and even get a preview of the moving... but when the task finishes, nothing was moved??? What???

This is because Draft Move allows users to start the task with an unmovable item selected (the padding), even though it can't move the item when the moving is tried applied by the user - and as no feedback is given, the newbie has no idea why the object (item's parent) wasn't used. This isn't made better that the user is even served with a preview of a move of the unmovable item's parent, the actual object.

Solution: the user should never be allowed to start a Draft Move task with an incorrect item - feedback should be given to why the task was denied.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-01:
If changes are made to Draft Move then please update the documentation as well https://wiki.freecad.org/Draft_Move

Roy-043, 2023-09-04:
Some unmovable objects can be copied. What the user wants to do, move or copy, depends on inputs in the task panel.

Identifying PD features is doable see:
https://github.com/FreeCAD/FreeCAD/blob/41a1a7a8e6256cc785ef6aff4efbe208e796eb2b/src/Mod/Draft/draftfunctions/downgrade.py#L196-L212

Roy-043, 2023-09-05:
Forum topic reporting a related issue:
https://forum.freecad.org/viewtopic.php?t=80928
Program hangs/crashes when trying to Draft_Rotate an attached sketch in a PD Body.

Roy-043, 2023-09-05:
Question: If a feature from a PartDesign Body is selected, maybe the whole Body should then be moved?