FreeCAD/FreeCAD #10472: [Problem]

ozzyman, 2023-09-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
[code]
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git) Snap 759
Build type: Release
Branch: tag: 0.21.1
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.7, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/Australia (en_AU)
Installed mods: 
  * kicadStepUpMod 10.19.4
  * CfdOF 1.21.7
  * OpticsWorkbench 1.0.9
  * CfdOF.backup1691301479.913904 1.21.7 (Disabled)
  * OSE3dPrinter
  * freecad.gears 1.0.0
  * sheetmetal 0.2.63
  * fasteners 0.4.59
  * 3D_Printing_Tools
  * POV-Ray-Rendering
  * Lithophane
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

Similar to issue #0900. I'm not using any external workbench however.

Filleting the indicated edge causes a hard crash. Just selecting the edge in the fillet dialog also causes an instant crash.

Steps to reproduce:

1. Open the attached part in FreeCAD
2. Select Part Design workbench
3. Select the edge (or any edge on the rear face of either slot)
4. Click the fillet tool
5. Crash

Selecting the fillet tool and then the edge also results in a crash. 

### Anything else?

[image]

[part.zip](https://github.com/FreeCAD/FreeCAD/files/12492023/part.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-01:
> Similar to issue https://github.com/FreeCAD/FreeCAD/pull/900. I'm not using any external workbench however.  

How is 900 related ?

ozzyman, 2023-09-01:
#9000

luzpaz, 2023-09-01:
9000 is a dupe of #8179

grubertechnic, 2023-09-01:
#9000 and #8179 the stacktrace looks pretty different.

in that case, it is more duplicate from #8179, the stacktrace is the same.

```
Err: Program received signal SIGSEGV, Segmentation fault.
Err: #0  /lib/x86_64-linux-gnu/libc.so.6(+0x3bfd0) [0x7ff48c25afd0]
Err: #1  0x7ff4340d2b64 in BRep_Tool::CurveOnSurface(TopoDS_Edge const&, opencascade::handle<Geom_Surface> const&, TopLoc_Location const&, double&, double&, bool*) from /lib/x86_64-linux-gnu/libTKBRep.so.7+0x164
Err: #2  0x7ff4340d2c3a in BRep_Tool::CurveOnSurface(TopoDS_Edge const&, TopoDS_Face const&, double&, double&, bool*) from /lib/x86_64-linux-gnu/libTKBRep.so.7+0xaa
Err: #3  0x7ff4340df8a5 in BRepAdaptor_Curve2d::Initialize(TopoDS_Edge const&, TopoDS_Face const&) from /lib/x86_64-linux-gnu/libTKBRep.so.7+0xd5
Err: #4  0x7ff3a833ea70 in ChFi3d_Builder::StartSol(opencascade::handle<ChFiDS_Spine> const&, opencascade::handle<BRepAdaptor_Surface>&, gp_Pnt2d&, opencascade::handle<BRepAdaptor_Curve2d>&, double&, opencascade::handle<ChFiDS_SurfData> const&, bool, int, opencascade::handle<BRepAdaptor_Surface>&, opencascade::handle<BRepAdaptor_Curve2d>&, bool&, bool&, bool&, bool&, opencascade::handle<BRepAdaptor_Surface>&, gp_Pnt2d&, bool, TopoDS_Vertex const&) const from /lib/x86_64-linux-gnu/libTKFillet.so.7+0xc40
Err: #5  0x7ff3a8343e64 in ChFi3d_Builder::PerformSetOfSurfOnElSpine(opencascade::handle<ChFiDS_ElSpine> const&, opencascade::handle<ChFiDS_Stripe>&, opencascade::handle<BRepTopAdaptor_TopolTool>&, opencascade::handle<BRepTopAdaptor_TopolTool>&, bool) from /lib/x86_64-linux-gnu/libTKFillet.so.7+0x674
Err: #6  0x7ff3a8346728 in ChFi3d_Builder::PerformSetOfKGen(opencascade::handle<ChFiDS_Stripe>&, bool) from /lib/x86_64-linux-gnu/libTKFillet.so.7+0xf8
Err: #7  0x7ff3a834b785 in ChFi3d_Builder::PerformSetOfSurf(opencascade::handle<ChFiDS_Stripe>&, bool) from /lib/x86_64-linux-gnu/libTKFillet.so.7+0x65
Err: #8  0x7ff3a8313395 in ChFi3d_Builder::Compute() from /lib/x86_64-linux-gnu/libTKFillet.so.7+0x415
Err: #9  0x7ff3a8491b7d in BRepFilletAPI_MakeFillet::Build(Message_ProgressRange const&) from /lib/x86_64-linux-gnu/libTKFillet.so.7+0x2d
Err: #10  0x7ff3a579c67d in PartDesign::Fillet::execute() from /home/user/Programs/FreeCADmaster/build/Mod/PartDesign/_PartDesign.so+0x71d
Err: #11  0x7ff48e8b27c2 in App::DocumentObject::recompute() from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADApp.so+0x92
Err: #12  0x7ff3a929326b in Part::Feature::recompute() from /home/user/Programs/FreeCADmaster/build/Mod/Part/Part.so+0xb
Err: #13  0x7ff48e85a09c in App::Document::_recomputeFeature(App::DocumentObject*) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADApp.so+0x28c
Err: #14  0x7ff48e861711 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADApp.so+0x571
Err: #15  0x7ff48e916164 in App::DocumentPy::recompute(_object*) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADApp.so+0x2c4
Err: #16  0x7ff48e916273 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADApp.so+0x23
Err: #17  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x1c9ba1) [0x7ff48dbc9ba1]
Err: #18  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x7d) [0x7ff48db7be8d]
Err: #19  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x4a63) [0x7ff48db09c63]
Err: #20  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x3ccd3c) [0x7ff48ddccd3c]
Err: #21  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyEval_EvalCode+0xc1) [0x7ff48dc68891]
Err: #22  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x2b7359) [0x7ff48dcb7359]
Err: #23  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyRun_StringFlags+0x6b) [0x7ff48dcb850b]
Err: #24  0x7ff48e32db78 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADBase.so+0x58
Err: #25  0x7ff48f412e90 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADGui.so+0x80
Err: #26  0x7ff48f412fec in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADGui.so+0xcc
Err: #27  0x7ff48f4130f2 in Gui::Command::updateActive() from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADGui.so+0x32
Err: #28  0x7ff3a4f14c85 in finishFeature(Gui::Command const*, App::DocumentObject*, App::DocumentObject*, bool, bool) from /home/user/Programs/FreeCADmaster/build/Mod/PartDesign/PartDesignGui.so+0x235
Err: #29  0x7ff3a4f19212 in finishDressupFeature(Gui::Command const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Part::Feature*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, bool) from /home/user/Programs/FreeCADmaster/build/Mod/PartDesign/PartDesignGui.so+0x652
Err: #30  0x7ff3a4f22ba8 in makeChamferOrFillet(Gui::Command*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) from /home/user/Programs/FreeCADmaster/build/Mod/PartDesign/PartDesignGui.so+0xb8
[... cut ...]
Err: #63  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7ff48c246285]
Err: #64  /home/user/Programs/FreeCADmaster/build/bin/FreeCAD(+0x9d11) [0x55ee524bad11]
```

```
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34104 (Git)
Build type: Release
Branch: master
Hash: db4a4fabc2144ac44575685fbf83531a68477e3e
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
```

grubertechnic, 2023-09-02:
Should not that e closed as duplicate?

closed 2023-09-02 by luzpaz