FreeCAD/FreeCAD #10479: [Assembly] User can't move/rotate parts easily

PaddleStroke, 2023-09-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
Latest 0.22
```

### Subproject(s) affected?

Other (specify in description)

### Problem description

In most CADs when you are working on an assembly, you can move parts of the assembly by clicking and dragging. This let you :
- easily position parts roughly at the beginning
- Move parts in a constrained assembly to make 'real' motions.

short video from OnShape (3:04), I think it will make it clearer what I mean : https://www.onshape.com/en/resource-center/videos/build-your-first-assembly

It is also usual to be able to rotate parts by holding ctrl while clicking and moving mouse.

Forum discussion :
https://forum.freecad.org/viewtopic.php?t=80773

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2023-09-02:
This should also include "natural" boundaries or constraints like:

- no intersection of solid bodies
- prevent bodies from being stretched or compressed when linked to other objects (e.g. hinges)

PaddleStroke, 2023-09-04:
That would be ideal but it's probably be pretty difficult to implement.

PR #10764 (2024-02-13): [Assembly] Solve the assembly (Card 9).
Adds AssemblyObject and viewer in cpp that handle the solving by connecting to ondsel solver.
Fixes #10753
Fixes #10749
Fixes #10754
Fixes #10479
Fixes #11311
Fixes #10751
Fixes #11397
Fixes #10815
Fixes #11606
Fixes #10752
closed 2024-02-13 by chennes