FreeCAD/FreeCAD #10480: [Assembly] has no insert fasteners tools

PaddleStroke, 2023-09-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Full version info

```shell
0.22 dev
```

### Subproject(s) affected?

Other (specify in description)

### Problem description

Assembly 4 has a 'Insert Fasteners' that is an integration of the Fasteners wb in a group command. This is very convenient to build assemblies.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-09-01:
@shaise are you willing to bring the fastener workbench in core assembly in the form of a toolbar concentrating functionality ?

grubertechnic, 2023-09-01:
Please no database addon in a core assembly. Very inconvenient when using it commercial, since pretty every company have its own database (folder, sql, pdm/plc, erp, however).

A general insert from function would be appropriate, which can be adapted.

PaddleStroke, 2023-09-01:
@grubertechnic There is already a general insert tool that insert links. Can you please check it out to tell me if this is what you had in mind as a general insert ?

I would argue that it's not because it would be in core that you would have to use it. People who want to use their own library can do so with the general insert tool no?

grubertechnic, 2023-09-02:
A insert from a library to be explicit. And you can add and define more then one source library. For example a sql, folder, https/ftp, script ..... whatever.

PaddleStroke, 2023-09-04:
Ok so I suggest we add a 'insert fasterner' tool similar to the one in A4 (so using the Fasterner wb database). But to make it possible to customize/replace the libraries in the assembly settings.

grubertechnic, 2023-09-05:
> using the Fasterner wb database

Default should be empty.

tobiasfalk, 2023-09-15:
> > using the Fasterner wb database
> 
> Default should be empty.

I do not agree with this, the default should be the FreeCAD internal(core) one, but I fully agree that the ability to add one's own database amd creation tool for Fasteners is defenetly a good idea. 
The reason why an empty default value/tool is a bad idea ist that it will create "wrong" errors from users that first start using the WB ant think that the button is not working becaus nothing is happening.

tobiasfalk, 2023-09-21:
A tool like this for inserting Fasteners in different shapes and sizes is defenetly a must have but I think it would also be nice if it would expand to other different standard parts like bearings, hinges or so on.
In this way I would call it a insert standard part, where one category is Fasteners.

grubertechnic, 2023-09-25:
> A tool like this for inserting Fasteners in different shapes and sizes is defenetly a must have but I think it would also be nice if it would expand to other different standard parts like bearings, hinges or so on.
> In this way I would call it a insert standard part, where one category is Fasteners.

You know, most companies have their own libary, even when they are only standart parts, because there are often much much more information stored because they are needed.

example: simple bold ISO4014 M12x60, and then?

From standardization details like, coating (bluing, phosphating, standard zinc plated (A2F), other non invasive zinc plating, chrome plating, ......), property class (8.8, 10.9, 12.9, XX-70, XX-80), material (A2, A4), ....

To individual details like part number (for BoM, PDM, ERP), accessibility (stock type, non stock type), commonness (very uncommon part, uncommon part, common part), mass, flags (quality management, receipt, .....), alternatives (ISO4017, DIN931, DIN933 (they are different to ISO, but more more common), ......

example: roller bearing, even their size can be defined due DIN616 (or other) norms, some details have a certain degree of freedom. For example the lubing groove in a spherical roller bearing is in SKF and FAG not the same size. Looks not important, but it is when you have no fixed position on a loose bearing side due length tolerances (machining, assembling, axial thrust (position moving on fixed bearing side) temperature, ..... ). Also the fillet on the edges can have some variations and they are important if a bearing really fits on a seat or just gets a line contact on the fillet edges.

So here a only write about the size and dimensions, when it gets on the real details (cage, roller numbers, materials, coatings, pre installed sealings (and their pre lucubration in the bearing), then you will see, that is damn individual.

Besides a pre installed library is, at least for me as frequent cad user, damn annoying, it does not really fits everybody need. Also from my experience even on small workshop shacks they have often they own library because many properties are very individual, even they are standardization parts. Maybe no ERP or PDM, but still very individual.

howie-j, 2023-09-25:
@grubertechnic what are you trying to say? That because large companies has spent hundreds/thousands of hours developing their own fasteners and standard parts library, FreeCAD should not include default fasteners? That's BS.

A standard insert fastener feature should at least be able to specify standard (e.g ISO 4762), dimension (M8x60), property (8.8) and material or coating (A2-70, zink plated etc). Then you have a proper standard part (without a MPN, that is what makes it a standard part) with enough information for any purchase department to be able to purchase the correct one. If you need to give each individual fastener a custom part number, manufacturer part number etc then you save that part to your PDM or ERP system, and adds the details there. When you then want to add that part to an assembly, you use the standard "insert part" command, not the "add standard fastener" command.

So yes, the default assembly should definitely come with the most used fasteners by default, and it should be fairly easy to add custom fasteners and standard parts to the library. Luckily enough the fasteners workbench already has the most used fasteners available, so the challenge here is to add a nice way of inserting the fasteners into an assembly.

grubertechnic, 2023-09-25:
> That because large companies has spent hundreds/thousands of hours developing their own fasteners and standard parts library, FreeCAD should not include default fasteners?

No, that is not what i say. As i said, even small shacks (as mine) have their individual standard parts, because they are adapted to their needs. They create their standard parts on the run and have no ERP data in it, but still individual.

Also to break it down, how do you consider to implement it as default? Per internet when FreeCAD is default internet opt in (default out)? Or implemented the very huge library in the installer to get it work offline? And what is the standard appearance?

similar minimum example a bolt ISO4014

[image]

[image]

appearance
 - with or without circle plane
 - with or without chamfers
 - with or without simplified thread markings (and when with, with simplified thread depth that it is the same as a nut or real thread depth)
 - with our without outer cut circle diameter on the hexagon
 
As you see, only the simple bolt can have a various but still valid appearance level of detail, which some users like or do not like. Implement all will not work. And even for a small companies side, that will not work at all, contrariwise it will trouble when you use the wrong implemented library per accident (as i have seen IRL in other companies, small and big, not speaking from mine).

defini-tiv, 2023-10-20:
Wouldn't it make sense to have a empty user-library by default. But with an obivious "add" button wich can import specific parts from a standard library.

So you would have one button click more for eveey part you use the first time. But it forces you organize your parts better. It will be much faster when you have addded most of the parts that you use more often.

maxwxyz, 2024-02-16:
Personally I am in favor of the Fasteners WB because this is a great starting point for personal use. In general a component library would be great, either customizable or maybe other addons can "dock" onto the Assembly WB and display a button/menu entry to open the corresponding component library.

Zolko-123, 2024-09-03:
> **Problem** description
> 
> Assembly 4 has a _Insert Fasteners_ that is an integration of the Fasteners wb in a group command. This is very convenient to build assemblies.

in what way is this a "_problem_ " ?

cbenhagen, 2025-05-11:
Corresponding issue in Fasteners WB: https://github.com/shaise/FreeCAD_FastenersWB/issues/416

PaddleStroke, 2026-04-18:
Updated prototype : 

https://github.com/user-attachments/assets/a1bbd7df-3c9a-406d-a21e-ee83458b7603

NXTler, 2026-04-18:
I agree. We should keep the fastener tool somewhat modular in preparation for other standard parts, so we don't end up in a situation where we have multiple tools for similar functions.