FreeCAD/FreeCAD #10482: [Regression] Sketcher: circular references in expressions go undetected

mfro0, 2023-09-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If I remember correctly, there once was a restriction in Sketcher that prevented references to elements in the same Sketch. This appears to have been removed in one of the more recent FreeCAD versions (you can now easily make your rectangle twice as high as wide).
There appears to be no check for circular references, however (so you now can make your rectangle twice as high on every recalculation, which doesn't seem to be right).

### Full version info

```shell
[code]
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34104 (Git)
Build type: Release
Branch: master
Hash: db4a4fabc2144ac44575685fbf83531a68477e3e
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * lattice2 1.0.0
  * Silk 0.1.4
  * sheetmetal 0.2.63
  * Help 1.0.3
  * Assembly4 0.50.2
  * Help.backup1661594724.9387012
  * Behave-Dark-Colors.backup1661598364.718598
  * Assembly4.backup1661591923.0362408
  * DynamicData 2.46.0
  * Curves 0.6.13
[/code]
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PLChris, 2023-09-01:
In revision 24291 I get the correct message "Reference constraint from this sketch cannot be used in this expression". Hope this text will help finding the regression.

Syres916, 2023-09-02:
Having looked up @PLChris message, then that is in https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Sketcher/App/SketchObject.cpp#L8448 Having checked back to 2019, only one line has changed other than formatting :

`    assert(expr != 0);`

to:

`    assert(expr);`

So it could be deeper in the expression engine possibly.

Syres916, 2023-09-03:
The function that does the cyclic dependency check is https://github.com/FreeCAD/FreeCAD/blob/master/src/App/PropertyExpressionEngine.cpp#L529 and that has changed quite a bit since 2019. So I now have something to debug hopefully.

Syres916, 2023-09-03:
Narrowed it down to the last 3 months of 2022 so running a Git Bisect, probably take a few hours including my sleep.

Syres916, 2023-09-03:
So https://github.com/FreeCAD/FreeCAD/commit/9498691ea8254b8214a4c9d4b243da358e16b597 created the regression in Sketcher so just need to find a way to capture both the Sketcher and Placement etc cyclic references without breaking something else.

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31501 (Git)
Build type: Release
Branch: (no branch, bisect started on releases/FreeCAD-0-21)
Hash: 9498691ea8254b8214a4c9d4b243da358e16b597
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Assembly4 0.50.2
  * Silk 0.1.3
  * A2plus
  * Defeaturing 1.2.0
  * CurvedShapes 1.0.4
  * MakerWorkbench 1.0.1
  * freecad.gears 1.0.0
  * SearchBar 1.0.2
  * FreeCAD_Turning_Addon 0.2.0
  * Curves 0.6.13
  * Help 1.0.3
  * toSketch 1.0.1
  * AirPlaneDesign 0.4.0
  * fasteners 0.4.56
  * sheetmetal 0.2.63

```

Syres916, 2023-09-03:
It's pretty ugly but seems to work https://github.com/FreeCAD/FreeCAD/compare/master...Syres916:FreeCAD:Sketch_Constraints_Cyclic_Ref

@wwmayer I like to know your thoughts please, I'm sure there's a more polished way to achieve this. I can also look at enhancing the Sketcher expression test to check this in future.

wwmayer, 2023-09-03:
> It's pretty ugly but seems to work

Well, it's bad practice to reference things of application modules. Such code is very fragile and will break as soon as things change in that application module. However, it's rather unlikely that the property will be renamed but it's still bad.

Before changing anything in the code I will start to write some unit tests...

bgbsww, 2023-12-09:
So, after a lot of code reading and debugging, as well as going through the forum thread and its reference to https://github.com/FreeCAD/FreeCAD/commit/14dfae059742a I wrote some tests, and I'm not sure I see any reason ever to null out the propDep.  But I could definitely be missing some cases.  I'd appreciate some commentary of other examples to test to prove this works.  https://github.com/FreeCAD/FreeCAD/compare/main...bgbsww:FreeCAD:bgbsww-patch-21  @wwmayer

bgbsww, 2023-12-12:
@luzpaz would you assign me, please?  Also, I'm running pretty low on HelpWanted Bugs with Milestones.  Suggestions are welcome if there are hotspots that need attention.

PR #11716 (2024-01-07): Detect circular references in sketches, and add corresponding tests
Fix #10482.
closed 2024-01-07 by chennes