FreeCAD/FreeCAD #10488: Arrow down on ComboView sometimes makes my cursor move all the way to the top.

leoheck, 2023-09-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Recently (when 0.21 was still to be released) this started to happen.

I usually click somewhere on the ComboView, and then most of the time I use the arrow down to navigate, and this moves the selection all the way to the top. Now, once it is on top, I can click anywhere in the combo view and the arrow down starts to work properly. This may be related to the removal of the Description column, maybe. 

I also found a way to replicate this when the cursor is working... If something is selected in the combo view and you hit the right arrow once, and then the down arrow, the selection moves to the top in the same fashion.

### Full version info

```shell
OS: Ubuntu 23.04 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33992 (Git) AppImage
Build type: Release
Branch: master
Hash: 90bafb3a069468455d8c65d030ef8750b510c39b
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * WebTools
  * Defeaturing 1.2.0
  * Curves 0.6.13
  * fasteners 0.4.63
  * lattice2 1.0.0
  * freecad.gears 1.0.0
  * sheetmetal 0.2.63
  * Assembly4 0.50.2
  * kicadStepUpMod 10.19.4
  * Curves.backup1692246817.490895 0.6.12 (Disabled)
  * ProDarkThemePreferencePack 1.0.0
  * Help 1.0.3
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-09-02:
Can't confirm this Tree view scroll behavior.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33981 (Git)
Build type: Release
Branch: master
Hash: 5a153e50ffb6046b6cc4dba498344032f6edb952
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

Syres916, 2023-09-02:
@leoheck can you please confirm that all patches for Ubuntu 23.04 are applied, this isn't the first time that the gui has been responsive for strange behaviour and the patches were already available.

leoheck, 2023-09-03:
@Roy-043 

My problem is with the selection. 

What I do is to click on one item in the tree to select something then I will do the rest with the arrow keys moving up/down the selection. 

If I just click the on item of the list, it is selected. Then when I hit arrow down to navigate downwards, the selected element jumps to the root of the tree. If I click one element in the list again, the arrow down starts to work again.

I could also observe that I can replicate the same behavior when I hit right arrow first (something like moving the selection to the next colum that doesn't exist anymore). And then this same behavior starts happening again. Now I could also see that, if I hit left arrow first (like moving the selection back to the first column or something, since the selection still remains in the same first column), the odd behavior is fixed. 

I could make a screenrecord but I am not sure how I can display the hits on the keyboard. I will have to check if I can do that.

@Syres916 sure, but what are the patches are you talking about? I upgrade my system multiple times a day, every day.

leoheck, 2023-09-03:
Ah, then this Left-Arrow happens when I want to unfold a Body/Part/Folder. And I also have to hit Left-Arrow twice to unfold each of these things.

There is more, when the selection is on the root, I cannot move it down by hitting Down-Arrow, I have to do Right-Arrow to unlock it. 

https://github.com/FreeCAD/FreeCAD/assets/1277920/a75057cd-cdf6-4531-a5cc-682bd6065fda

leoheck, 2023-09-03:
This behavior is a bit annoying, but at least I would find a workaround while reporting this issue. The workaround is to use the Left/Right arrows to fix something to be able to use the Up/Down arrows again.

leoheck, 2023-11-30:
I saw this issue got the `Has workaround` tag, what is the workaround? This is an annoying issue.

Roy-043, 2023-12-04:
@zlice This is not a Help forum. Please start a discussion in the [forum](https://forum.freecad.org/) for this.

@leoheck You yourself have proposed a workaround.
> The workaround is to use the Left/Right arrows to fix something to be able to use the Up/Down arrows again.

leoheck, 2023-12-04:
Ah, sure my workaround, alright. I thought it was the workaround to fix the issue.

Roy-043, 2023-12-05:
@zlice Your issue is different. If you create a GitHub issue be sure to add the necessary details. You message here is lacking in that regard.

zlice, 2024-11-25:
This seems worse in 1.0.0

luzpaz, 2024-11-25:
@zlice can you add your full [About](https://wiki.freecad.org/About) info please? TIA

zlice, 2024-11-25:
Sorry for the poor video, just trying to make it small.

- Click item
- press right-arrow (nothing)
- use other tree with right-arrow (nothing)
- use mouse to expand
- able to go down/up (this didn't work my previous try but did this time)
- able to collapse with left-arrow
- now able to expand with right-arrow
- but now left-arrow or down move sporadically wherever they want

If anything stands out as obviously wrong please let me know.

```
OS: Void Linux (Shynebox/xcb)
Architecture: x86_64
Version: 1.0.0.1_voidlinux
Build type: Release
Python 3.12.7, Qt 6.7.2, Coin 4.0.0, Vtk 9.3.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/qt6ct-style
Installed mods: 
  * Manipulator 1.5.0
  * Curves 0.3.0
  * kicadStepUpMod
```

https://github.com/user-attachments/assets/8f773e84-6a3b-468b-b5a2-00b642b8190a

leoheck, 2024-11-25:
> This seems worse in 1.0.0

Yeah, this behavior is horrible.
I think it is related with the hidden Description column. Do you have this column visible or hidden?

zlice, 2024-11-25:
Took me a second to find it. Indeed `right click > tree settings > description` makes things much more bearable.

commit c7620fd9 (2025-01-24): Fix arrow navigation in TreeWidget

Closes #10488. This commit removes the code fragment that "swallowed"
the key press events.
closed 2025-01-24 by chennes