FreeCAD/FreeCAD #10494: [Problem] Part design, sketcher, pocket doesn't work with specific bspline case

benj5378, 2023-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Can't create pocket with this very specific sketch, I suspect it has to do with the bspline
[M12.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/12503994/M12.FCStd.zip)

https://github.com/FreeCAD/FreeCAD/assets/12400097/740968d3-318d-4f0e-94ac-64495c80dc9c

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34091 (Git) Snap 758
Build type: Release
Branch: master
Hash: a2735416e1189566329b327d99f115ced7071b86
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-09-02:
You create 2 self intersections in the sketch (on both sides).

[image]

Therefore no bug or issue.

benj5378, 2023-09-02:
Thanks, it was the issue. Is this meant to not throw an error when trying to perform a pocket?

grubertechnic, 2023-09-02:
Even a very good CAD kernel can crap out on near tangents or near co-tangential co-face operations. Some more, some less, i sort such behavior as normal.

grubertechnic, 2023-09-02:
tbh, a candidate to close.

benj5378, 2023-09-02:
If a second opinion agrees, that a pocket feature that doesn't make a pocket shouldn't throw a warning or an error, then let's close it.

luzpaz, 2023-09-04:
@AjinkyaDahale (of bspline fame) care to weigh-in ?

AjinkyaDahale, 2023-09-04:
From the behaviour it looks like an OCC bug. Please try with a larger cross section if you can for your specific case.

As for closing the issue, I would prefer to keep a stock of "known issue" cases where the software breaks in the math. Closing this issue may imply to the reader that the problem is solved.

grubertechnic, 2023-09-05:
There are already many cases, where a defect pad is not detect correctly. Take any bop part or part design issue or helix operation.

AjinkyaDahale, 2023-09-06:
Then maybe go ahead and mark this as a "duplicate" of the closest one.

benj5378, 2025-03-24:
Reopen

luzpaz, 2025-03-26:
> Then maybe go ahead and mark this as a "duplicate" of the closest one.

@benj5378 is this a dupe ?

benj5378, 2025-03-26:
I'm not sure this is a duplicate. Some also state, that the fail is due to OCCT.

While it's true it is OCCT error, then this issue is not about fixing that error. This issue proposes throwing an error in FreeCAD, instead of failing silently