FreeCAD/FreeCAD #10495: [Feature request] Draft, Draft Move, if axis keys held down, apply axis lock temporarily

benj5378, 2023-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, if `x`, `y` og `z` keys are tapped during a Draft Move, the moving will be locked to only the selected axis. First part of video.

Now, if the keys are held down, a lot of lag occurs, as the function is made for tapping, not continuous key pressing. Second part of video.

Preferably, these two types of keys presses should be distinguished. If axis keys are tapped, the axis locks should be apply until axis keys are tapped again. If axis keys are held down, the axis lock should be applied (with no lag) until the keys are released again.

https://github.com/FreeCAD/FreeCAD/assets/12400097/66d8c90b-f935-4f3d-8749-e2dba9b4044b

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34091 (Git) Snap 758
Build type: Release
Branch: master
Hash: a2735416e1189566329b327d99f115ced7071b86
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-09-02:
We already have the Shift key that can be held down.

yorikvanhavre, 2023-09-04:
That's the difference between a modifier key (shift, ctrl, alt), that can be hold pressed without emitting new signals constantly, and a normal key. When you keep a normal key pressed, it emits the signal repetitively and AFAIK there is no way in Qt to use a normal key as a modifier key (I researched that too a while back)

Indeed you can press Shift to obtain the same result, so probably we can consider this issue as solved...

I'll close this issue because I don't think we can do more unfortunately... Please reopen if you think this should not be closed

benj5378, 2023-09-04:
I hoped to at least fix the glitch if wrongly holding down the `x`, `y` and `z` keys. But I do see Qt is the issue here

closed 2023-09-04 by luzpaz

benj5378, 2023-09-04:
> When you keep a normal key pressed, it emits the signal repetitively and AFAIK there is no way in Qt to use a normal key as a modifier key (I researched that too a while back)

@yorikvanhavre If I find a way to distinguish if a key is pressed or kept down, would it be relevant to make the shift key obsolete, such that: if axis keys are tapped, the axis locks should be apply until axis keys are tapped again, if axis keys are held down, the axis lock should be applied (with no lag) until the keys are released again?

yorikvanhavre, 2023-09-05:
Yes, if those x, y or z keys could be used as a shift key, that would of course be excellent, and I don't see why we wouldn't implement this. But I also looked into this already, and didn't find anything useful...

benj5378, 2025-04-05:
> That's the difference between a modifier key (shift, ctrl, alt), that can be hold pressed without emitting new signals

I had a look at this, and it seems the issue stems from the key checking. It seems pressing `x` is detected by checking if the QLineEdit had a textEdit event where `x` was the new character. Using `QWidget::keyPressEvent` and `QWidget::keyReleaseEvent` it should be possible to check if held down or not. Though it's very difficult to change with the current implementation using QLineEdit::textChanged signal.