FreeCAD/FreeCAD #10496: [Problem] Draft, Draft Move axis lock ignores global

benj5378, 2023-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, if `x`, `y` og `z` keys are tapped during a Draft Move, the moving will be locked to only the selected axis... if we are using the working plane as our coordinate system reference. First part of video, notice of `Local Y` and `Local Z` stays `0`.

If I tick off `Global` the global coordinate system should be used, and I also expect axis locks to be applied to the global axis. Other than this being expected behaviour and useful, the purpose of `Global` tick can be questioned (from a functional point) if it doesn't do anything, when doing axis locking.

But this is not the case: axis locks are still applied to the working plane coordinate system, and this should be changed. Second part of video, see how `Global Y` and `Global Z` are all over.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34091 (Git) Snap 758
Build type: Release
Branch: master
Hash: a2735416e1189566329b327d99f115ced7071b86
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-09-02:
Confirmed. This should be fixed.

benj5378, 2023-09-02:
This also applies to using `shift` key. Furthermore, `shift` key doesn't work with z axis.

Roy-043, 2023-09-04:
I have done a quick test implementation and encounter an issue that I first thought was a coding error, but is really due to how lines are projected in the 3D view.

When constraining to an axis the current point is projected perpendicularly onto a line parallel to that axis. If the axis is not parallel to the Working Plane (and also not more or less perpendicular to the view direction) The resulting angles can be confusing. See images below.

[image]
[image]

Steps to reproduce:
1. Place the attached `gui_snapper.py` file in `/Mod/Draft/draftguitools/`
2. Restart FreeCAD.
3. Switch to Std_ViewIsometric.
4. Switch to the Draft WB.
5. Call Draft_SelectPlane and select "Align to view"
6. Call Draft_Line.
7. Click the first point of the line.
8. Make sure the "Global" checkbox is checked.
9. Press the "X" key once.
10. Move the cursor around.

[gui_snapper.zip](https://github.com/FreeCAD/FreeCAD/files/12512890/gui_snapper.zip)

PR #11436 (2023-11-18): Draft: constraining did not respect global mode
Fixes #10496.
commit f6f0ace4 (2023-11-18): Draft: constraining did not respect global mode

Fixes #10496.
closed 2023-11-18 by Roy-043