FreeCAD/FreeCAD #10497: [Problem] Draft, meaning of global confusing

benj5378, 2023-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Now, let's imagine a newbie user, that haven't experienced distinguish between local and global coordinate systems before. They are wondering what the purpose of global is - and if it isn't checked, then what coordinate system is used, and what is the local coordinate system?

The labels should be more descriptive, first and foremost, it should be `(Use) global coordinates` rather than just `Global`. Secondly, it should be clear what is used instead, either global coordinates or working plane coordinates; tick boxes are used when "apply this or nothing", not when "apply this or that instead" (global or working plane), then radio buttons should be used instead, and this would make this much clearer: -  `(Use) global coordinates` - `(Use) working plane coordinates`. Sure the tooltip shows this, but this would make it much better.

A discussion to whether global coordinates should be raised. I have never ever needed/preferred using working plane coordinates, and I doubt it being different from other users; having working plane coordinates as default only adds a good amount of confusion to the top of how Draft tools' coordinates work.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34091 (Git) Snap 758
Build type: Release
Branch: master
Hash: a2735416e1189566329b327d99f115ced7071b86
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-09-02:
Global is a normal term in CAD. See no issue here. Even through CAD is just "software" it is no plaything and must be learned from new users or used from authorities. This is also the reason why commercial CADs have with a new license also some training lessens included.

Roy-043, 2023-09-02:
The Working Plane is a key concept of the Draft WB. Therefore local coordinates are the default. This should not be changed IMO.

Using radios instead of a checkbox is a good idea though.

benj5378, 2023-09-02:
> Global is a normal term in CAD

100%. But there is no reason to have `global` instead of `global coordinates`, sympatheticly understood, incomprehensible for new users

benj5378, 2023-09-02:
> The Working Plane is a key concept of the Draft WB. Therefore local coordinates are the default. This should not be changed IMO.

By key concept, do you mean important feature or mostly used? Important features isn't necessarily more used.

@Roy-043 So just to clarify; when you use Draft Move, you use it with working plane the majority of the time?

grubertechnic, 2023-09-02:
> 100%. But there is no reason to have `global` instead of `global coordinates`

The word global inherent in its context global coordinate system.

> incomprehensible for new users

As many many points in CAD in general. As i said, CAD is no plaything and must be learned. It is no stupid gatekeeping, it is inherent on its circumstances. 

An analogy: a programmer must learn what is `private`/`public`, `classes`, mutates, variable types, namespaces, and so on, independent on the programming language. Most words here are more weird (example git) and still must learn it.

benj5378, 2023-09-02:
> As many many points in CAD in general. As i said, CAD is no plaything and must be learned. It is no stupid gatekeeping, it is inherent on its circumstances.
>
> An analogy: a programmer must learn what [...]

I don't really get why this is relevant. So you say that you want it to be `global` instead of `global coordinates`, just so people will have to learn it? What you're saying is, that because CAD is not a plaything, it _has_ be difficult to learn.

grubertechnic, 2023-09-02:
> So you say that you want it to be `global` instead of `global coordinates`, just so people will have to learn it?

No. I mean, even you spell it out, it would you not help you either when you do not get the context, the extra word would not help here.

> What you're saying is, that because CAD is not a plaything, it _has_ be difficult to learn.

No, it _is_ not easy to learn. Go on a lathe and make a fit. You will not succeed when you are not learned. Try to buy the tools on the "thing you stuck the drill" on a lathe. When do you not know, that is a tailstock (however this name was generated) and the cone is a morse taper, you will get nothing. CAD is in your language fagarbeider work, want it or not. 

When you write everything out, it is good only for the first moment when you are new. Then when you know it, it annoys on the gui, because it need too much space.

benj5378, 2023-09-02:
> When you write everything out, it is good only for the first moment when you are new. Then when you know it, it annoys on the gui, because it need too much space.

This is a valid discussion point (though I disagree on it).

grubertechnic, 2023-09-02:
A small suggestion on it, a mouseover function with a few seconds over the label, where it is spelled out.

Roy-043, 2023-09-02:
> when you use Draft Move, you use it with working plane the majority of the time?

I am currently experimenting with code related to the Draft Working Plane, and therefore using it a lot. But that has very little statistical relevance. The same applies to your personal experience with the Draft WB. And let's face it, the 3 people participating in this discussion so far are hardly 'average' users anyway.

> mouseover function with a few seconds over the label

There is a tooltip already.

grubertechnic, 2023-09-02:
> And let's face it, the 3 people participating in this discussion so far are hardly 'average' users anyway.

True, this is the reason, why something like this should "fight out" in the forum, to write martial, see https://forum.freecad.org/viewtopic.php?t=80864.

benj5378, 2023-09-03:
> I am currently experimenting with code related to the Draft Working Plane, and therefore using it a lot. But that has very little statistical relevance. The same applies to your personal experience [...]

You're saying that you don't know what the average users uses the most... but then why do you think working plane should be the default (and not global), just because it's a key feature - if you're saying it hasn't to do what people uses most?

Roy-043, 2023-09-04:
The Working Plane is almost literally the foundation of the Draft WB. When picking random points you are picking points on the Working Plane. Even if you have switched the display of the coordinates to Global. So Local coordinates are really the most sensible default.

yorikvanhavre, 2023-09-04:
Setting global as a default instead of local is possible, if a lot of users ask for it, as long as it's possible to change I have no problems with it. But removing useful features (I for one use WP coords a lot) to make FreeCAD "easier" would seem very wrong to me.

I'd agree with radio buttons, but it makes the UI less compact too, and honestly I am not sure that would make things so much "easier"? Basically if someone doesn't know the difference between global and local coords, this won't help.

benj5378, 2023-09-04:
Comments here and including some Discord talk seems to show, that I am the only one who find any of the things I raise an issue in regards to wording and radio buttons.

Regards to what should be the default, I don't really get the relevance of the arguments behind that the default shouldn't be global, just because it doesn't work like that under the hood. But it seems like I am in the minor too.

So let's consider closing this.

Roy-043, 2023-09-05:
OK, closing as requested.

closed 2023-09-05 by Roy-043