FreeCAD/FreeCAD #10510: [Problem] Draft, task panel, checkbox and buttons can't be interacted with using keyboard

benj5378, 2023-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using `enter` key while having selected a button (with `tab` key) should be equivalent to clicking the button.
Using `space` key while having selected a checkbox (with `tab` key) should be equivalent to clicking the checkbox.

But neither `enter` or `space` keys work on buttons or checkboxes inside the task panel. Instead using space hides the object selected in the hierarchy (even though it hierachy isn't selected)?

So far, this only seems to affect task panels in Draft

### Full version info

```shell
OS: Fedora Linux 38 (KDE Plasma) (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4.backup1691951820.225657 0.50.2
  * Assembly4
  * Behave-Dark-Colors 0.1.1
  * Curves 0.6.13
  * DynamicData 2.46.0
  * fasteners 0.4.59
  * Freecad-Built-in-themes-beta 1.2.2
  * freecad.gears.backup1692223246.5590403 1.0.0
  * freecad.gears
  * ExtremeProLight 1.6.0
  * A2plus 0.4.60n
  * ExplodedAssembly
  * FreeCAD_DarkPreferencePAck 0.0.1
  * Glass
  * kicadStepUpMod 10.19.4
  * Silk 0.1.4
  * lattice2 1.0.0
  * Manipulator 1.5.0
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

benj5378, 2023-09-02:
Confirmed by @qewer33

yorikvanhavre, 2023-09-04:
I guess we need to handle these key events at button or checkbox level otherwise they are passed upwards...

benj5378, 2023-09-04:
@yorikvanhavre any ideas why this only seems to affect Draft Workbench?

luzpaz, 2023-09-04:
This also falls in to the 'accessibility' category

luzpaz, 2023-09-28:
@Roy-043 when you get a chance, would you mind weighing-in?

Roy-043, 2023-09-29:
In my tests on Windows toggles and buttons on Draft task panels can be accessed via the keyboard.

I have used Draft_Line and Draft_ShapeString in my tests.

`Space` works for toggles.
`Enter` or `Space` works for buttons (Draft_ShapeString) or only `Space` (Draft_Line).

Steps:
1. Start FreeCAD.
2. Start a new document.
3. Switch to the Draft WB.
4. Call one of the mentioned commands.
5. Click in the X inputbox.
6. Press `Shift+Tab` twice. The Cancel button has the focus now.
7. Press `Enter` or `Space`.

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34387 (Git)
Build type: Release
Branch: master
Hash: d95b26dfb5dd23eaee004b4af3cf5cf4766e7f9c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

benj5378, 2025-01-24:
This is still an issue. Only some tasks are affected, like Move and Rotate

luzpaz, 2025-01-26:
@maxwxyz not sure why the 'Accessibility' tag was removed. We want FC to be also be utilized for folks with handicaps (physical disadvantages)