FreeCAD/FreeCAD #10512: [Problem] Cannot uninstall Help Add-on

CAKCy, 2023-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Trying to uninstall Help Add-on freezes the program on a window with title "Keep addon settings?"

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3 (Disabled)
[/code]
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-09-03:
Confirmed.

Roy-043, 2023-09-03:
It is possible to delete the addon manually:
1. In the Python console enter `App.getUserAppDataDir()`.
2. Go to that folder, and then to the `Mod` subfolder.
3. Remove the `Help` folder from the `Mod` folder.

Syres916, 2023-09-03:
I wonder if on Windows the other small dialog box title Removing Addon with a Cancel button is hidden behind the other dialog box. On Linux I was able to Cancel and then click No to the Keep addon settings one and therefore remove successfully with a minor warning which I don't remember seeing before:

`QBackingStore::endPaint() called with active painter; did you forget to destroy it or call QPainter::end() on it?`

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Assembly4 0.50.2
  * Silk 0.1.3
  * A2plus
  * Defeaturing 1.2.0
  * CurvedShapes 1.0.4
  * MakerWorkbench 1.0.1
  * freecad.gears 1.0.0
  * SearchBar 1.0.2
  * FreeCAD_Turning_Addon 0.2.0
  * Curves 0.6.13
  * toSketch 1.0.1
  * AirPlaneDesign 0.4.0
  * fasteners 0.4.56
  * sheetmetal 0.2.63
```

CAKCy, 2023-09-03:
> I wonder if on Windows the other small dialog box title Removing Addon with a Cancel button is hidden behind the other dialog box. On Linux I was able to Cancel and then click No to the Keep addon [...]

Thanks for your reply!
I tried moving the other small window out of the way and then clicked on OK. The issue persists.

chennes, 2023-09-03:
@yorikvanhavre that is your dialog, I believe. Any ideas? Do we have conflicting modals between the Addon Manager and your uninstall script?

yorikvanhavre, 2023-09-04:
I don't remember honestly... I don't think anything special to handle modal order had been done there.

Now, this is a good reminder for something else, the Help module should not be removable. In fact, I think it should become built-in, as there is now no more other way to access the documentation. I'll open a separate issue

yorikvanhavre, 2023-09-06:
Do you mean this page? https://wiki.freecad.org/Online_Help_Startpage
Indeed I think it needs a heavy refactoring :sweat_smile: 
The Help addon now uses the online version by default, so this message is not accurate anymore...

CAKCy, 2023-09-06:
> Do you mean this page? https://wiki.freecad.org/Online_Help_Startpage Indeed I think it needs a heavy refactoring 😅 The Help addon now uses the online version by default, so this message is not accurate anymore...

😄

yorikvanhavre, 2023-09-08:
Ok I did a good update to that page... I think most of it was still @jriegel 's version from one or two decades back ;)

CAKCy, 2023-09-08:
> Ok I did a good update to that page... I think most of it was still @jriegel 's version from one or two decades back ;)

On behalf of myself and future generations I thank you! 👏

yorikvanhavre, 2023-09-11:
Opened an issue for merging the Help module at https://github.com/FreeCAD/FreeCAD/issues/10527
This issue should be closed once #10527 is implemented

luzpaz, 2023-09-23:
@yorikvanhavre please add `Fixes #10512` in to the PR that fixes #10527, TIA!

PR #11008 (2023-10-16): Merging the Help module
Merging the help module from https://github.com/FreeCAD/FreeCAD-Help 

fixes #10527
fixes #10512
commit 1c187686 (2023-10-16): Merging the Help module (#11008)

* Adding the Help module - fixes 10527, fixes #10512

* Aded Help to pre-commit

* [pre-commit.ci] auto fixes from pre-commit.com hooks

for more information, see https://pre-commit.ci

---------
closed 2023-10-16 by chennes