FreeCAD/FreeCAD #10514: [Problem] Failed to compute left/right minimum bearings for "cursor.pcf"

ville-h, 2023-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If I try to use the provided "Dark theme" or the "Light theme" preference packs and then try to open the "view > panels > report view" freecad starts reporting: `Failed to compute left/right minimum bearings for "cursor.pcf"` repeatedly. Eventually consumes 100% of a core and becomes unresponsive.

I also tried getting other themes with "tools > addon manager" and they exhibited the same behaviour.

Reason I'd like to use the "Dark theme" is that otherwise I'll have white text on white background in the spreadsheet rendering it unusable. Should this be reported separately?

### Full version info

```shell
[code]
OS: Arch Linux (openbox)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: makepkg
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.11.5, Qt 5.15.10, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-09-04:
@MisterMakerNL have you encountered this before ?

luzpaz, 2023-09-05:
@ville-h what version of fontconfig is Arch using ?

ville-h, 2023-09-05:
On my system:
```
extra/fontconfig 2:2.14.2-1 [installed]
```

cfillion, 2023-09-11:
Replacing `<FCText Name="Font">cursor.pcf</FCText>` to `monospace` in ~/.config/FreeCAD/user.cfg under `<FCParamGroup Name="Editor">` seems to have fixed it for me. The log is no longer displayed in a huge font, and FreeCAD no longer freezes in an infinite loop of "Failed to compute left/right minimum bearings" errors.

ville-h, 2023-09-11:
Based on cfillion's message decided to take a closer look. Running `find / -name '*cursor.pcf' 2>/dev/null` on my system doesn't produce any results. Running `find / -name '*cursor.pcf.gz' 2>/dev/null` does. Removing `.pcf` and using `<FCText Name="Font">cursor</FCText>` works. So few observations:

0) Using a name like: `cursor.pcf` rather than `cursor` your themes rely on the font file being in an specific format on disk

1) Your font loading mechanism doesn't seem to handle errors very well. It stubbornly retries the same thing over-and-over resulting in problems

InfiniteCoder01, 2023-09-15:
I've got this issue on FreeCAD link daily 0.21.0 2023.08.11.
I'm on NixOS with Hyprland. When starting, I also get this issue: `Could not find the Qt platform plugin "wayland" in ""`

> Replacing `<FCText Name="Font">cursor.pcf</FCText>` to use `monospace` in ~/.config/FreeCAD/user.cfg under `<FCParamGroup Name="Editor">` seems to workaround the issue for me. The log is no longer displayed in a huge font, and FreeCAD no longer becomes unresponsive under the load of infinite "Failed to compute left/right minimum bearings" warnings.
> 
> Before (it's that big despite the size being 10 in user.cfg): [image]
> 
> After (correct display + doesn't hang): [image]

This doesn't help. I don't even have `<FCParamGroup Name="Editor">`, only `<FCParamGroup Name="Editor" />`

ville-h, 2023-09-15:
> Should we just disallow cursor.pcf as a possible font option ?

Whilst that's something to consider it's really not a very solid fix. At the very least you may want to effect a rule on every font name that they can't reference the filename extension directly and thus depend on the on-disk format. But event that would just paper over the more acute problem that a failure with your font loading mechanism is able to effectively lock up the whole program. Which is bound to re-manifest because of something else at a later date if not fixed.

I've not looked at the code, but I am guessing what ends up happening is that the "report view" uses the particular font named "Font" and every time it tries to display text it sees if the font is loaded. And every time that font fails to load it will try to display an error message about the font failing to load which causes it try to load the font...

What you may want to have is some sort of rate limiting mechanism on the error messages. You could for example hash the error message and store a "last seen" timestamp in a hash table. This way you can stop such a runaway error chain from happening. It's not very useful to spam the "report view" full of the same error over and over anyways.

adrianinsaval, 2023-09-21:
Where did you get these preference packs from? I could not find any reference to "cursor.pcf" in our code

ville-h, 2023-09-21:
I've simply installed the official arch linux freecad [package](https://archlinux.org/packages/extra/x86_64/freecad/). I did also try the aur freecad appimage [package](https://aur.archlinux.org/packages/freecad-appimage). Both exhibited the same problem, but at the time I tried the appimage-one I didn't realize a user's preference file was storing the font name.

However if I go and edit the font name in the preference file the change/fix seems to get overwritten if I go into "edit > preferences" and click "ok" with out changing anything. So it keeps getting "cursor.pcf" from somewhere.

adrianinsaval, 2023-09-25:
in a recent PR another problematic font got blacklisted, perhaps we should do the same with this cursor.pcf font: https://github.com/FreeCAD/FreeCAD/pull/10555

FredM7, 2023-11-04:
I thought that I should mention that I get the same error when I go to `General` and change `Unit System` to `MKS (m, kg, s, degree)`. That is when the same error shows up for me - and then FreeCad's process runs at ~27% with the `Memory usage` rapidly climbing. 

If i change it to `Standard (mm, kg, s, degree)` it seems to go away.

One problem for me is that once I have changed it to MKS, it becomes non-responsive, and even after a restart the problem persists. The only way for me to get rid of it is to just simply delete my entire `user.cfg` file, mainly due to a lack of knowing which specific value to manually remove in the config file, at the moment.

I'm using NixOS 23.11 - FreeCad version 0.21.1

adrianinsaval, 2023-11-05:
are you able to reproduce on 0.22?

Andrei-Pozolotin, 2023-11-14:
the error message is from the bowels of QT5:
https://github.com/qt/qtbase/blob/5.11/src/gui/text/qfontengine.cpp#L653

the error perhaps indicate a need to fix the use of QT font fallback mechanism
since in the long term there will be no end to fonts that will require black listing

Andrei-Pozolotin, 2023-11-14:
archlinux `cursor` font is present in the system:
```
/usr/share/fonts/misc/cursor.pcf.gz
```

as provided by the default package:
https://archlinux.org/packages/extra/any/xorg-fonts-misc/

and looks "normal":
```
fc-list | grep cursor
/usr/share/fonts/misc/cursor.pcf.gz: cursor.pcf:style=Regular
```

Andrei-Pozolotin, 2023-11-14:
with FreeCAD v 0.22 dev build from current master the above `user.cfg` hacks do not work: 
any change in
    `FreeCAD -> Menu -> Edit -> Preference -> Application -> Theme -> [change here]`
results in `user.cfg` being re-writen with
    `<FCText Name="Font">cursor.pcf</FCText>` 
config entry restored and the application hangs again

Andrei-Pozolotin, 2023-11-14:
hey, a better hack: use a launch script, to always have a good font after hang :-)
```bash
#!/usr/bin/env bash
config_file="$HOME/.config/FreeCAD/user.cfg"
/usr/bin/sed -i 's/>cursor.pcf</>monospace</g' $config_file
exec /usr/bin/FreeCAD
```

luzpaz, 2023-11-26:
This seems like it's Linux specific. Can we get more traction on this ticket ?

Andrei-Pozolotin, 2023-11-29:
@efferre79 : try this, then OK, then restart.
```
FreeCAD -> Edit -> Preferences -> Python -> Editor -> Display Items -> Text -> Family -> [SOME OTHER FONT]
```

this works for me (may be more):
* Liberation mono
* DejaVu mono
* Monospace
* Ubunto mono

Andrei-Pozolotin, 2023-11-29:
> I cannot find any reference in user.cfg about the font Courier

there are some 9 places in the source where "Courier" is hard coded as a default for missing font preference value

efferre79, 2023-11-29:
> @efferre79 : try this, then OK, then restart.
> 
> ```
> FreeCAD -> Edit -> Preferences -> Python -> Editor -> Display Items -> Text -> Family -> [SOME OTHER FONT]
> ```
> 
> this works for me (may be more):
> 
>     * Liberation mono
> 
>     * DejaVu mono
> 
>     * Monospace
> 
>     * Ubunto mono

I followed your instruction selecting DejaVu Sans mono and the problem disappeared. 

But it's interesting that the font settings found there (when I opened first time the dialog) was using 8514fix even if was not present in the config file. To recap, when I start without user.cfg, freecad is looking for Courier font, then as soon as I open the editor dialog there is some automatic guessing functionality (in freecad or most likely in Qt) that autoselect the font 8514fix (at least on my system). Then by closing the dialog with apply/ok, without touching anything, the bad 8514fix is stored in the user.cfg file.

efferre79, 2023-11-29:
In my opinion a safety check must be implemented to avoid this kind of situation because, starting from a clean config file, the problem might arise if on the system there are fonts incompatible with Qt font-engine code

Andrei-Pozolotin, 2023-11-29:
> a safety check must be implemented to avoid this kind of situation

Sure. There is actually some logic in the code which is supposed to take care of cases like that. I am tracing the origin of the problem now - no dice so far.

Andrei-Pozolotin, 2023-11-30:
@luzpaz : mystery solved:

PROBABLE PROBLEM:

* all fonts are good, but `QFontEngine` thinks otherwise

* when `QFontEngine::minRightBearing()` logic fails, it logs a "bad font" `qWarning()`:
https://code.qt.io/cgit/qt/qtbase.git/tree/src/gui/text/qfontengine.cpp?h=5.15#n652

* since `FreeCAD` feeds console output back into the GUI `PythonConsole`, `ReportView`, etc, this eventually calls into QT layout manager which asks `QFontEngine` for font sizes, which trips on "bad" fonts and produces next `qWarning()`, which feeds back to the console, which ..., resulting in repeated stack traces like this:
```
QFontEngine::minRightBearing
QTextLine::layout_helper
QPlainTextDocumentLayout::layoutBlock
QPlainTextDocumentLayout::blockBoundingRect
QTextCursorPrivate::blockLayout
QTextCursorPrivate::movePosition
QTextCursor::movePosition
Gui::PythonConsole::printPrompt
Gui::PythonConsole::PythonConsole
Gui::MainWindow::setupPythonConsole
```
* that feedback loop is what consumes 100% process CPU time, as reported above

* when "Courier" exact match font family is asked for, but is not present on the OS system, (again, according to `QFontEngine`), QT will automagically find the "next best thing", which could also happen to be a `QFontEngine` "bad" font, such as `cursor.pcf`,`8514oem`,`8514fix`, and many more, depending on the system

* there are more then 9 places in FreeCAD which use "relaxed but hard-coded" "Courier" font family discovery, such as:

`/FreeCAD/src/Gui/PreferencePages/DlgSettingsEditor.cpp`
```cpp
QFont getMonospaceFont()
{
    QFont font(QString::fromLatin1("monospace"));
    if (font.fixedPitch())
        return font;
    font.setStyleHint(QFont::Monospace);
    if (font.fixedPitch())
        return font;
    font.setStyleHint(QFont::TypeWriter);
    if (font.fixedPitch())
        return font;
    font.setFamily(QString::fromLatin1("courier"));
    if (font.fixedPitch())
        return font;
    return font; // We failed, but return whatever we have anyway
}
```

or no discovery at all, such as:

`/FreeCAD/src/Gui/PythonConsole.cpp`
```cpp
    QFont serifFont(QLatin1String("Courier"), 10, QFont::Normal);
    setFont(serifFont);
```

* when default `QApplication::font()` is "bad" in `QFontEngine` judgement, all these places are candidates to blow up

POSSIBLE SOLUTIONS:

* so perhaps, one would want a single centralized "fixed font family factory" which always gives "good fonts"

* one could ensure that default `QApplication::font()` is setup really early and is a really "good" font

* preference filtering code could be extended to ensure "good fonts":

`/FreeCAD/src/Gui/PreferencePages/DlgSettingsEditor.cpp`
```cpp
void DlgSettingsEditor::loadSettings()
    for (const auto &name : familyNames) {
        if (QFontDatabase().isFixedPitch(name)) {
            if (name.compare(QLatin1String("8514oem"), Qt::CaseInsensitive) != 0) {
                fixedFamilyNames.append(name);
            }
        }
    }
```

with logic such as this (tested, seems to work):
```python
def check_has_good_font_family(family_name:str) -> bool:
    # https://code.qt.io/cgit/qt/qtbase.git/tree/src/gui/text/qfontengine.cpp?h=5.15#n245
    # #define kBearingNotInitialized std::numeric_limits<qreal>::max()
    kBearingNotInitialized = -2147483648
    from PySide2 import QtGui
    this_font = QtGui.QFont()
    this_font.setFamily(family_name)
    this_metrics = QtGui.QFontMetrics(this_font)
    if this_metrics.minRightBearing() == kBearingNotInitialized:
        return False  # produces: qWarning() << "Failed to compute left/right minimum bearings for" << fontDef.family;
    elif this_metrics.minLeftBearing() == kBearingNotInitialized:
        return False  # produces: qWarning() << "Failed to compute left/right minimum bearings for" << fontDef.family;
    else:
        return True
```

* and, finally, one could also replace the "Courier" with whatever is the hip font of today

**We failed, but return whatever we have anyway®** 😍

efferre79, 2023-11-30:
@Andrei-Pozolotin thanks for the digging in the code

I think that filtering is the way to go also to identify only the fixed pitch fonts which Qt likes. On my system the following python code works, of course it has to be translated in C++ and inserted in freecad code. But one more step is needed to filter out the qWarning() messages during the scan to avoid flooding the system command line and the report View window with useless messages for the user (I am sure it is possible to disable and enable somehow during the scan).

If the font suggested by getMonospaceFont() is not found, there is no problem, the first good font coming from the filtering is picked up. A check is needed of course if no good fonts from the filtered list is found.

```
import PyQt5.QtGui as g
import PyQt5.QtWidgets as w
import sys

app=w.QApplication([])
db=g.QFontDatabase()

for fam in db.families():
  if db.isFixedPitch(fam):
    f=g.QFont(fam)
    m=g.QFontMetricsF(f)
    if m.minRightBearing() != sys.float_info.max and m.minLeftBearing() != sys.float_info.max:
      is_ok=True
    else:
      is_ok=False
    print("%s: %d" % (fam,is_ok))
```

The filtering of 8514oem in current code was added [here](https://github.com/FreeCAD/FreeCAD/issues/10122). Maybe there is a more general rule to remove other fonts like that in the filtering procedure.

Andrei-Pozolotin, 2023-11-30:
@luzpaz:

 some modules bring their own fonts:
`/FreeCAD/src/Mod/TechDraw/Gui/AppTechDrawGui.cpp`
```cpp
void loadTechDrawResource()
    std::string fontDir = App::Application::getResourceDir() + "Mod/TechDraw/Resources/fonts/";
    std::vector<std::string> fontsAll(
        {"osifont-lgpl3fe.ttf", "osifont-italic.ttf", "Y14.5-2018.ttf", "Y14.5-FreeCAD.ttf"});
    for (auto& font : fontsAll) {
        QString fontFile = Base::Tools::fromStdString(fontDir + font);
        int rc = QFontDatabase::addApplicationFont(fontFile);
```

perhaps `FreeCAD` proper could afford a default "branded font" as well

efferre79, 2023-12-01:
Besides in `src/Gui/PreferencePages/DlgSettingsEditor.cpp`, I see other places where the QFontDatabase is used to gather all system  fonts, in particular:

```
src/Gui/propertyeditor/PropertyItem.cpp
src/Gui/PreferencePages/DlgSettingsNavigation.cpp
```

I don't know if the same issue can appear also there, but if that is the case it would make sense to have an unique point in freecad code where to scan (and filter) fonts.

luzpaz, 2023-12-02:
CC @WandererFan JFYI, maybe you want to keep this issue on your radar as it is Font related

Andrei-Pozolotin, 2023-12-10:
how this issue is manifested on `windows`: https://github.com/FreeCAD/FreeCAD/issues/11681

luzpaz, 2023-12-10:
Upstream is asking for a 'minimal example', can you provide one? 

ref: https://bugreports.qt.io/browse/QTBUG-119690?focusedId=761926&page=com.atlassian.jira.plugin.system.issuetabpanels%3Acomment-tabpanel#comment-761926

Andrei-Pozolotin, 2023-12-11:
@luzpaz : 

while working on the 'minimal example', I found a solution for the initial font discovery, as follows:

**1) replace the resolver code**

https://github.com/FreeCAD/FreeCAD/blob/ac7a409d1ddf4a03d3a9e4ec952d0aedf94d5dd0/src/Gui/PreferencePages/DlgSettingsEditor.cpp#L58

with: (ignore `qWarning()`)
```cpp
bool hasFixedPitch(const QFont &font) {
   const QFontInfo font_info(font); // <<== CRITICAL QFontInfo
   const auto family_name = font_info.family();
   const auto has_fixed_pitch =  font_info.fixedPitch();
   qWarning() << "### hasFixedPitch():" << family_name << has_fixed_pitch;
   return has_fixed_pitch;
}

QFont getMonospaceFont() {
    QFont font; // default QApplication font
    font.setStyleHint(QFont::Courier); // may not work
    if (hasFixedPitch(font)) {
        return font;
    }
    font.setFamily(QString::fromLatin1("Monospace")); // may not work
    if (hasFixedPitch(font)) {
        return font;
    }
    font.setFamily(QString::fromLatin1("Courier")); // may not work
    if (hasFixedPitch(font)) {
        return font;
    }
    font = QFontDatabase::systemFont(QFontDatabase::FixedFont); // should always work
    if (hasFixedPitch(font)) {
        return font;
    }
    qWarning() << "### getMonospaceFont(): failure";
    return font;
}
```

**2) and replace the resolution result**
https://github.com/FreeCAD/FreeCAD/blob/ac7a409d1ddf4a03d3a9e4ec952d0aedf94d5dd0/src/Gui/PreferencePages/DlgSettingsEditor.cpp#L297
with
```cpp
QByteArray defaultMonospaceFont = QFontInfo(getMonospaceFont()).family().toLatin1(); // <<== CRITICAL QFontInfo
```

**3) why:**

there are 3 ways to ask `.isFixedPitch()`, `.family()`, etc:

```python
def has_fixed_pitch_1(this_font:QtGui.QFont) -> bool: # GOOD
    return QtGui.QFontDatabase().isFixedPitch(this_font.family())

def has_fixed_pitch_2(this_font:QtGui.QFont) -> bool: # GOOD
    return QtGui.QFontInfo(this_font).fixedPitch()

def has_fixed_pitch_3(this_font:QtGui.QFont) -> bool: # NOT GOOD
    return this_font.fixedPitch()
```

the basic error is the confusion of `QFont` vs `QFontInfo` and `QFontDatabase`:
one way to stay "unconfused" is to remember that:
* `QFont` is like http request message
* `QFontInfo` is like http response message
* `QFontDatabase` is like http server

current specific `DlgSettingsEditor` code error stems from:
`QFont.fixedPitch()` reflects only result of `QFont.setFixedPitch()`
`QFont.family()` reflects only result of `QFont.setFamily()` or of constructor `QFont(...)`
in other words, no resolution is used at all

perhaps more people need to test this proposed fix on multiple OS

other problems mentioned above still remain:
* font filtering
* prevent feedback loop
* full font-related code review

Andrei-Pozolotin, 2023-12-11:
@luzpaz : 

added first  'minimal example':
https://bugreports.qt.io/browse/QTBUG-119690?focusedId=762541&page=com.atlassian.jira.plugin.system.issuetabpanels:comment-tabpanel#comment-762541

Andrei-Pozolotin, 2024-05-09:
@sliptonic:

this build exhibits the same bug on the `archlinux`:
* `Ondsel_ES_2024.2.0.37191-Linux-x86_64.AppImage` https://ondsel.com/blog/ondsel-es-2024-2/

please let me know if you will have time to review and accept a PR for the proposed solution above:
* https://github.com/FreeCAD/FreeCAD/issues/10514#issuecomment-1849176386

polygon, 2024-06-21:
> Replacing `<FCText Name="Font">cursor.pcf</FCText>` to use `monospace` in ~/.config/FreeCAD/user.cfg under `<FCParamGroup Name="Editor">` seems to workaround the issue for me. The log is no longer [...]

This only fixes things temporarily for me. As soon as I change something in the Preferences Dialog (which I do regularly to change detail level for STL exports), the faulty setting is being restored and also immediately applied to the current session, so it starts using all CPU and I got only a few seconds left before the program becomes completely unresponsive.

> Whilst that's something to consider it's really not a very solid fix.

It's almost 9 months now without any fix. Maybe something that is not rock-solid but un-breaks the program would be in order?

maxwxyz, 2024-07-01:
@MisterMakerNL @ville-h  is this still relevant?

ville-h, 2024-07-01:
No idea. I stopped using freecad.

Andrei-Pozolotin, 2024-07-02:
@maxwxyz : 
* yes, the "font error" issue is still there
* this proposed change (`getMonospaceFont()` code in the comment) will resole the issue:
https://github.com/FreeCAD/FreeCAD/issues/10514#issuecomment-1849176386
* please assign someone to review and accept a PR for the proposed change that I will submit readily

maxwxyz, 2024-07-02:
just tag me in the PR

luzpaz, 2024-07-30:
@Andrei-Pozolotin you've been assigned. Thanks for your work here. We look forward to the PR.

Andrei-Pozolotin, 2024-07-30:
ok

jl1990, 2024-09-10:
This seems to be still happening. Is there any update on when this will be fixed?

Thanks a lot!

maxwxyz, 2024-09-13:
@Andrei-Pozolotin any news

deftdawg, 2024-10-27:
I'm still having this issue with the latest 1.0RC2 from nixpkgs unstable :cry: 
```
ls -larthd /nix/store/*freecad*
dr-xr-xr-x 9 root root 4.0K Dec 31  1969 /nix/store/g1nwhca7h5vkhcd0w345zwz1jp0xz6kf-freecad-1.0rc2
-r--r--r-- 1 root root 9.5K Dec 31  1969 /nix/store/bw23sdj0czzhwwg2jm881rcjkv0v7dag-freecad-1.0rc2.drv
-r--r--r-- 1 root root 2.2K Dec 31  1969 /nix/store/al4r5ac6xh9xin9d7cdr489v74c3byg9-0003-freecad-font-issue-10514.patch
-r--r--r-- 1 root root 9.5K Dec 31  1969 /nix/store/939pnjrvb9awjdn0s7vbs4kzd9x9bnv8-freecad-1.0rc2.drv
```

The patch is present, but the intense spamming of this `Failed to compute left/right minimum bearings for "cursor.pcf"` message eventually crashes my freecad. 

 - system: `"x86_64-linux"`
 - host os: `Linux 6.11.3-xanmod1, NixOS, 24.11 (Vicuna), 24.11pre694395.4c2fcb090b1f`
 - multi-user?: `yes`
 - sandbox: `yes`
 - version: `nix-env (Nix) 2.24.9`
 - channels(root): `"nixos"`
 - channels(deftdawg): `""`
 - nixpkgs: `/nix/var/nix/profiles/per-user/root/channels/nixos`

```
cat /nix/var/nix/profiles/per-user/root/channels/nixos/.git-revision 
18536bf04cd71abd345f9579158841376fdd0c5a
```

hzeller, 2024-11-17:
With 9806a463246f5a0daf5d87b323ab8c06a97feba2 should this be marked fixed, or is it still happening ?

maxwxyz, 2024-11-17:
@ville-h @deftdawg is it still failing in RC4?

PassiveLemon, 2024-11-20:
I'm still getting this error. While it doesn't take up my entire screen anymore, it still outputs to stdout and occasionally pops up a message informing me that the error occured X times. Still get freezes too but I'm unsure if that's related to the cursor issue.

ville-h, 2024-11-23:
Yes I tested the recently added arch package for: `extra/freecad 1.0.0-1` and it's still happening. I no longer use freecad because of this.

evils, 2024-11-23:
i'm still getting this on sway on nixos-unstable with version 1.0.0

i can trigger it with the classic theme (default settings after removing `~/.local/share/FreeCAD` and `~/.config/FreeCAD`)
by going to preferences, changing nothing and pressing `apply`
that causes a spam and a lockup requiring a `pkill`

after that deftdawg's workaround seems to work
(though i still get the occasional `Failed to compute left/right minimum bearings for "cursor.pcf"`, there's no spam and it doesn't lock up)

applying any change in the preferences reintroduces the issue
(and the workaround then seems to work as well)

Andrei-Pozolotin, 2025-01-01:
@maxwxyz :

I can confirm that the commit: https://github.com/FreeCAD/FreeCAD/commit/9806a463246f5a0daf5d87b323ab8c06a97feba2 which implements a solution from above:  https://github.com/FreeCAD/FreeCAD/issues/10514#issuecomment-1849176386

resulted in the following changes in the behavior of this bug:

A) it does resolve the "feedback loop of death" issue when tested in this environment (no more hang):

```
OS: Arch Linux (KDE/plasmax11/xcb)
Architecture: x86_64
Version: 1.1.0dev.39748 (Git)
Build type: Release
Branch: makepkg
Hash: e977eb49e37ef836f695809263fa1f56a17cc657
Python 3.12.7, Qt 6.8.1, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/breeze
```

B) and while the burst of the logger entries in the report view:
```python
Failed to compute left/right minimum bearings for ...
``` 
does still occur during the font list discovery in `DlgSettingsEditor.ui`, on combo-box pull-down,
it does not result in the hang as before,  and this burst should be handled elsewhere, by the logging subsystem

dyfrgi, 2025-03-31:
Is there a workaround for this issue that's not "sed the config file and avoid ever opening the preferences dialog"?

Andrei-Pozolotin, 2025-04-01:
> I suspect it's the .pcf extension that is causing the issue on Linux.

No, it is not a `pcf`, and not any specific font, there are few already, and there will be more.

There is a long standing issue with QT font manager metrics code implementation.

When QT font manager stumbles on the font it "can not calculate",
it merely logs an error from private `cpp` code path, instead of raising exception,
and propagates invalid font metrics values back to the caller.

I see no way to work around it in a generic way, short of doing the following on FreeCAD startup:
* setup log interceptor on QT library
* trigger font manager font discovery 
* catch font metrics errors in the log, extract "bad" font names
* make sure FreeCAD is using only custom font dialog everywhere
* and remove "bad" font names from that custom font dialog on activation

epchris, 2025-04-22:
Just chiming in to say that I'm running into the same issue in Arch linux, I've tried several different ways of installing FreeCAD (app image, aur app image, aur weekly appimage, extras package) and am consistently running into the same issue.

cederom, 2025-05-19:
I have the same problem in FreeCAD 1.0.0 on FreeBSD 14.2-RELEASE AMD64 that shows up when importing STEP files and so workflow is completely blocked/broken :-(

eraserhd, 2025-05-22:
I have the same problem on NixOS with 1.0.0 and 1.0.1.

If I edit `user.cfg` so that it reads this:

```xml
          <FCText Name="Font">cursor</FCText>
```

instead of this:

```xml
          <FCText Name="Font">cursor.pcf</FCText>
```

The error goes away and I can use FreeCAD for more than ten minutes (it's now been running for several hours.  Before it would crash after creating a new document, making a couple rectangles in a new sketch, adding a few constraints -- I repeated these steps about five times.

Additionally, I no longer see the left/right warning, and it's clear that's the _only_ thing I did.  I chose dark mode when I first opened FreeCAD.