FreeCAD/FreeCAD #10517: FreeCAD not working with Windows 11

wieslawzuber, 2023-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After installing and starting FreeCAD, it shows the following error - The application cannot be started because MSVCP140_1.dll is missing

Po zainstalowaniu i uruchomieniu FreeCAD pokazuje taki błąd - Nie można uruchomić aplikacji bo brakuje MSVCP140_1.dll

### Full version info

```shell
The application cannot be started because MSVCP140_1.dll is missing
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-09-03:
Did you install 0.21.0, or 0.21.1? In either case the problem can be resolved by installing this package from Microsoft:
https://learn.microsoft.com/en-us/cpp/windows/latest-supported-vc-redist?view=msvc-170#visual-studio-2015-2017-2019-and-2022

wieslawzuber, 2023-09-03:
Thx ! Its working 😀

chennes, 2023-09-03:
Good: which version did you download?

wieslawzuber, 2023-09-03:
0.21.0

chennes, 2023-09-03:
Thanks: I believe I fixed this problem with the 0.21.1 installer.

Roy-043, 2023-09-04:
The mentioned file is indeed present in the 0.21.1 installer. Closing this.

closed 2023-09-04 by Roy-043