FreeCAD/FreeCAD #10523: chamfer cannot be created

gms2009, 2023-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On ubuntu 22 freecad 0.19

https://github.com/gms2009/FreeCAD/blob/master/ppp-99.FCStd
1 out of 4 chamfer cannot be created

https://github.com/gms2009/FreeCAD/blob/master/ppp-399.FCStd
4 out of 8 chamfer cannot be created

kind of hard to describe the 3d picture,   should be obious when open the files

### Full version info

```shell
ubuntu 22 freecad 0.19
release date 2022/02/14
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-09-03:
It is not obvious, there are many many edges in the files.

Besides the shapes are not redefined Also you seems to use an outdated version.

If there is not more refining the description and not repeatable in the 0.21 version, it is candidate to close the issue.

gms2009, 2023-09-03:
1 out of 4 cannot be created. so 3 are created
same as 4 out of 8

gms2009, 2023-09-04:
just compile 0.21.1 from source on ubuntu 22, same problem

Roy-043, 2023-09-04:
Confirmed. The chamfers that fails will work if the size is 0.4mm or less. This may be an OCC issue.

gms2009, 2023-09-04:
> It is not obvious, there are many many edges in the files.
> 
> Besides the shapes are not redefined Also you seems to use an outdated version.
> 
> If there is not more refining the description and not repeatable in the 0.21 version, it is candidate to close the issue.

what is " the shapes are not redefined" ?   how to redefine ?

Roy-043, 2023-09-04:
Most PartDesign features have a "Refine" property. Setting it to true will merge planar faces.

grubertechnic, 2023-09-05:
If the edges are meant under the V, then it is a bad sketch issue. There many weird constraints, see screenshot.

I have redone it with valid constraints, then it works flawless.

[image]

[ppp-399_edit.zip](https://github.com/FreeCAD/FreeCAD/files/12527573/ppp-399_edit.zip)

Please next time:
 - follow the issue rules, searching for you file to the correct edge, on a file with 131 edges is time consuming and therefore bearish.

 - if you need help, go in the forum, this is no helpdesk.

grubertechnic, 2023-09-05:
@Roy-043 : it think that is imho a candidate to close.

gms2009, 2023-09-06:
Sorry for that, I thought I found a bug, I just download the ppp-399_edit.zip, the problem is still there, the last 4 item are 4 chamfers they are edge180 edge60 edge78 edge94

gms2009, 2023-09-06:
and I need to create chamfer on edge41 190 181 6. right beside the 4 created, same shape same length, as Roy pointed out, the chamfer will work for size less than 0.4mm, that still a problem.

gms2009, 2023-09-06:
the picture shows looks like ppp-99 file, but the download link is ppp-399 nothing changed

grubertechnic, 2023-09-06:
i only edited the ppp-99 file and there is a new body, which is created by me. I do not bother the second file, it have the same constraint issues.

gms2009, 2023-09-06:
Looks like lots of mistakes in the early stages, I will be more careful.

closed 2023-09-07 by luzpaz