FreeCAD/FreeCAD #10524: [Problem] Sketcher code needs refactoring

FlachyJoe, 2023-09-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The sketcher workbench code needs a cleanup

- Gui/CommandConstraints.cpp is too big (10 000+ lines)
- quite same code is copied at three places for each constraints cases (in Gui/CommandConstraints.cpp , Gui/EditModeConstraintCoinManager.cpp and Gui/ViewProviderSketch.cpp)

### Full version info

```shell
[code]
OS: Debian GNU/Linux trixie/sid (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33981 (Git)
Build type: Release
Branch: master
Hash: 5a153e50ffb6046b6cc4dba498344032f6edb952
Python 3.11.5, Qt 5.15.10, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: French/France (fr_FR)
[/code]
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-09-04:
I was thinking a bit about the code structure of the commands actually.
I remember initially being very confused about where were commands, what were dsh and so on. Perhaps it would be best to have one file for each command.

Recently Abdullah split CommandCreateGeos because it was so huge. But he left CommandCreateGeo and just moved the DSH to individual files.
Perhaps it would be best to have individual files for commands : 
- CmdSketcherGeometryLine : include the command + the DSH
- CmdSketcherGeometryCircle
- CmdSketcherConstraintDistance
- CmdSketcherConstraintCoincident
- ...

But this also raise the question of uniformity of code between workbenches. Currently every workbench architecture is different, which makes it hard to discover. Maybe we should have structure rules that all workbenches have to follow. I think for contributors it would make life much easier. Which would result in more contributions and less confusion.

Rule example : 
- Commands should be in individual files with naming following the naming convention 'CmdWorkbenchName'. All commands files should be in the folder Gui/Commands. 
- The command files should contain all the code relative to this command (example : Taskbox, DSH...)

FlachyJoe, 2023-09-04:
> I would also add that it's not an issue, more like a discussion.

IMHO it's an issue for new contributors to understand the code and so it limits the possible improvement at user side.

> this also raise the question of uniformity of code between workbenches.

Code uniformity is a great target. We could at least write notice for developers to explicit the role of each file and how features are implemented i.e. a DevNote.md at each workbench top.

closed 2025-02-10 by github-actions[bot]