FreeCAD/FreeCAD #10526: [Problem] openDocument of Assembly3 document with freecadcmd results in `Aborted`

khimaros, 2023-09-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

i'm trying to set up CI/CD for my FreeCAD workflow. all of my parts individually can be opened and exported in headless mode using `freecadcmd` and a bit of python code.

however, when trying to open an Assembly3 document, `freecadcmd` aborts prematurely without any explanation as to why.

reproducible consistently with this bit of code:

```python
#!/usr/bin/env freecadcmd

import sys

import FreeCAD

path, dest = sys.argv[1:]

print('### opening document', path)

FreeCAD.openDocument(path)
```

the output is as follows:

```shell
FreeCAD 0.20.2, Libs: 0.20.2R
(c) Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2022
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

### opening document cad/atom/atom-assembly.FCStd
Aborted
```

### Full version info

```shell
[code]
OS: Debian GNU/Linux trixie/sid (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.20.2.
Build type: Release
Python 3.11.4, Qt 5.15.10, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * parts_library (Disabled)
  * sheetmetal 0.2.63
  * kicadStepUpMod 10.19.4
  * fasteners 0.4.61 (Disabled)
  * Assembly3 0.12.0
  * Manipulator 1.5.0
  * InventorLoader 1.4.0 (Disabled)
[/code]
```

### Subproject(s) affected?

Core

### Anything else?

this happens even with an empty assembly: [empty-assembly.FCStd.gz](https://github.com/FreeCAD/FreeCAD/files/12513786/empty-assembly.FCStd.gz)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

khimaros, 2023-09-04:
i filed the same issue over at Assembly3 here because i'm not sure which software is at fault: https://github.com/realthunder/FreeCAD_assembly3/issues/1319

Syres916, 2023-09-04:
I get a little more output:

```
QPixmap: Must construct a QGuiApplication before a QPixmap
Aborted
```

That tells me there's a `if FreeCAD.GuiUp:` missing from Assembly3 Wb somewhere to handle this scenario.

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34145 (Git)
Build type: Release
Branch: master
Hash: 0a8292a850b98d2e66e08e235dfb81b6e7285df1
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Assembly4 0.50.2
  * Silk 0.1.3
  * A2plus
  * Defeaturing 1.2.0
  * CurvedShapes 1.0.4
  * MakerWorkbench 1.0.1
  * freecad.gears 1.0.0
  * SearchBar 1.0.2
  * Assembly3 0.12.0
  * FreeCAD_Turning_Addon 0.2.0
  * Curves 0.6.13
  * toSketch 1.0.1
  * fasteners 0.4.56
  * sheetmetal 0.2.63
```

khimaros, 2023-09-04:
is this something i could workaround in my script? i tried using `Gui.setupWithoutGUI()` but didn't help

Syres916, 2023-09-04:
You're far better getting @realthunder 's input or open a topic in https://forum.freecad.org/viewforum.php?f=20 so more eye's are on it. I really don't think there's many users even trying this.

luzpaz, 2023-09-04:
Lets close this in favor of the downstream https://github.com/realthunder/FreeCAD_assembly3/issues/1319

closed 2023-09-04 by luzpaz

khimaros, 2023-09-06:
@Syres916 is there a good example of a workbench that handles this correctly that i could model some changes to Assembly3 on?

khimaros, 2023-09-06:
currently there are no checks for GuiUp in asm3:

```shell
$ grep -rn GuiUp ~/.local/share/FreeCAD/Mod/Assembly3/freecad/asm3/
```

luzpaz, 2023-09-06:
Recently i saw this PR: #10464